RIOT-OS/RIOT #2240: RPL_UDP: native (and SAMR21): fail build for RELEASE 2014.12

bapclenet, 2015-01-05:
The release 2014.12 fails to build for RDP_UDP example for native (and SAMR21 by removing it from the blacklist)

Here is the error:

    make BOARD=native all 
    Building application "rpl_udp" for "native" with MCU "native".
    /bin/sh: 1: Syntax error: "(" unexpected
    make[1]: ***     [/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/examples/rpl_udp/bin/native/rpl_udp/helper.o] Erreur 2
    make: *** [all] Erreur 2

LudwigOrtmann, 2015-01-05:
Not for me and it had not failed for Travis either. What's your toolchain, system, etc.? Also: `make info-build`?
Did you run `make clean`? What does `git status` say?

bapclenet, 2015-01-05:
I did a make clean and I downloaded the tar file of the release so no git status needed.
Everything worked fine with this conf before the release:

Toolchain:
<code>arm-none-eabi-gcc --version </code>
<code>arm-none-eabi-gcc (4.8.2-14ubuntu1+6) 4.8.2 </code>

System: 
<code>Linux version 3.13.0-32-generic (buildd@roseapple) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) )</code>

IDE: Eclipse Luna

<code>make BOARD=native info-build </code>
APPLICATION: rpl_udp

supported boards:
 * avsextrem
 * chronos
 * fox
 * iot-lab_M3
 * msb-430h
 * msba2
 * msbiot
 * native
 * redbee-econotag
 * samr21-xpro
 * telosb
 * wsn430-v1_3b
 * wsn430-v1_4
 * z1

BOARD:   native
CPU:     native
MCU:     native

RIOTBASE:  /home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec
RIOTBOARD: /home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards
RIOTCPU:   /home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/cpu

DEFAULT_MODULE: auto_init core cpu sys
DISABLE_MODULE: 
USEMODULE:      defaulttransceiver hashes ieee802154 nativenet net_help net_if periph posix ps routing rpl shell shell_commands sixlowpan socket_base timex transceiver uart0 udp vtimer

ELFFILE: /home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/examples/rpl_udp/bin/native/rpl_udp.elf
HEXFILE: /home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/examples/rpl_udp/bin/native/rpl_udp.hex

FEATURES_REQUIRED (excl. optional features):
         transceiver
FEATURES_OPTIONAL (strictly "nice to have"):
         -none-
FEATURES_PROVIDED (by the board or USEMODULE'd drivers):
         config cpp periph_cpuid periph_random periph_rtc transceiver
FEATURES_MISSING (incl. optional features):
         -none-
FEATURES_MISSING (only non-optional features):
         -none-

CC:      cc
CFLAGS: 
	-DDEVELHELP  
	-DFEATURE_TRANSCEIVER  
	-DMODULE_AUTO_INIT  
	-DMODULE_CORE  
	-DMODULE_CPU  
	-DMODULE_DEFAULTTRANSCEIVER  
	-DMODULE_HASHES  
	-DMODULE_IEEE802154  
	-DMODULE_NATIVENET  
	-DMODULE_NET_HELP  
	-DMODULE_NET_IF  
[... 89 lines cut ...]
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/fox/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/iot-lab_M3/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/mbed_lpc1768/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/msb-430h/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/msba2/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/msbiot/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/native/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/openmote/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/pca10000/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/pca10005/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/pttu/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/redbee-econotag/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/samr21-xpro/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/spark-core/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/stm32f0discovery/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/stm32f3discovery/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/stm32f4discovery/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/telosb/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/udoo/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/wsn430-v1_3b/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/wsn430-v1_4/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/yunjia-nrf51822/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/z1/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/boards/native/Makefile.features  
	/home/xsoen/Documents/Developpement/workspace/RIOT_Release_2014_dec/Makefile.vars

LudwigOrtmann, 2015-01-05:
Ah, you're not working on a git checkout.. I guess that's not well tested ;)
(Just out of curiosity - why?)

LudwigOrtmann, 2015-01-05:
I updated the issue title and description.

bapclenet, 2015-01-05:
I can test with a GIT checkout if you want but that shouldn't change anything :-), I let you know in a bit.
I wanted to see if the release uploaded works fine.

LudwigOrtmann, 2015-01-05:
The problem is this: ` -DRIOT_VERSION=\"UNKNOWN (builddir: /tmp/RIOT-2014.12)\"`. The shell fails on the bracket expression and it's a result of not knowing the git checkout.

This patch fixes the problem:
```diff
--- /home/lo/RIOT/Makefile.include	2015-01-03 17:26:51.221796611 +0100
+++ Makefile.include	2015-01-05 11:28:48.808365475 +0100
@@ -130,7 +130,7 @@
       RIOT_VERSION := $(GIT_STRING)-$(GIT_BRANCH)
     endif
   else
-    RIOT_VERSION := UNKNOWN (builddir: $(RIOTBASE))
+    RIOT_VERSION := "UNKNOWN (builddir: $(RIOTBASE))"
   endif
 endif
 export CFLAGS += -DRIOT_VERSION=\"$(RIOT_VERSION)\"
```

bapclenet, 2015-01-05:
It works, thank you :-)

bapclenet, 2015-01-05:
I close the issue

LudwigOrtmann, 2015-01-05:
no

LudwigOrtmann, 2015-01-05:
https://github.com/RIOT-OS/RIOT/pull/2241 needs merging first

bapclenet, 2015-01-05:
ok

LudwigOrtmann, 2015-01-05:
Thanks for trying our releases ;)

PR #2241 (2015-01-06): make: escape unknown version string
fixes #2240
closed 2015-01-06 by OlegHahm