RIOT-OS/RIOT #2247: core/lpm_arch: Clarification of the lpm arch interface

fnack, 2015-01-06:
I provided an implementation of the lpm_arch interface for the STM32F4 cpu (see #2246).

During that process, I wondered what the difference behind RIOT's LPM_POWERDOWN and LPM_OFF mode is. From my perspective, I couldn't find a difference and documentation doesn't help either. Maybe someone could explain the intention behind separating these two mode options?

If there is no difference, I'd vote for removing the LPM_OFF mode.

fnack, 2015-01-06:
Ping @haukepetersen. Here's the discussed new issue report from yesterday.

haukepetersen, 2015-01-06:
@OlegHahm do you remember the notion behind the different power levels?

OlegHahm, 2015-01-06:
No, these defines already existed when I started to work with FireKernel. @kaspar030?

saurabh984, 2015-01-06:
I've used these mappings for SAM D21 LPM if it helps!

```
LPM_ON = Run Mode
LPM_IDLE = SYSTEM_SLEEPMODE_IDLE_0 (Wake up source: Synchronous, (APB, AHB), asynchronous)
LPM_SLEEP = SYSTEM_SLEEPMODE_IDLE_1 (Wake up source: Synchronous (APB), asynchronous)
LPM_POWERDOWN = SYSTEM_SLEEPMODE_IDLE_2 (Wake up source: Asynchronous)
LPM_OFF = SYSTEM_SLEEPMODE_STANDBY (Wake up source: Asynchronous)
```

Datasheet Ref: http://www.atmel.com/Images/Atmel-42223-SAM-R21_Datasheet.pdf (Page: 115)

PeterKietzmann, 2015-01-06:
So as there are some devices whith multiple idle modes it seems to make sense that there are for example five different power modes. Correct? One could maybe rename them so it becomes clear that they differ in "strengh" but I don't mind staying with the existing solution.

gebart, 2015-01-08:
The Kinetis devices have even more power modes:

 - RUN
 - WAIT
 - STOP
 - VLPR
 - VLPW
 - VLPS
 - LLS
 - VLLS0
 - VLLS1
 - VLLS2
 - VLLS3

Having five different LPM enums is probably enough for most use-cases though.

haukepetersen, 2015-01-08:
I had a quick discussion with @OlegHahm about this. The most important transition seems to be the one called by the idle thread. In my opinion we could even go with less modes (as the kernel uses only IDLE and RUN), but put some platform specific code in place, so that each time the system goes into IDLE it will figure out the best mode available (depending on running peripherals and the hardware platform). Does this make sense to you?

PeterKietzmann, 2017-01-16:
Obsolete  (see #6160)

closed 2017-01-16 by PeterKietzmann