RIOT-OS/RIOT #2261: samr21-xpro: RTC clock runs significantly too slow

thomaseichinger, 2015-01-07:
After having the RTC run for an hour the diff already reached 1 and a half minutes.
@bapclenet any idea?

```bash
rtc settime 2015-01-07 15:19:00
2015-01-07 15:19:00,272 - INFO # > rtc settime 2015-01-07 15:19:00
2015-01-07 15:19:00,272 - INFO #
rtc gettime
2015-01-07 15:24:37,402 - INFO # > rtc gettime
2015-01-07 15:24:37,402 - INFO #
2015-01-07 15:24:37,440 - INFO # 2015-01-07 15:24:23

2015-01-07 16:22:36,812 - INFO # > rtc gettime
2015-01-07 16:22:36,813 - INFO #
2015-01-07 16:22:36,850 - INFO # 2015-01-07 16:20:03
```

bapclenet, 2015-01-07:
Hmm, weird, I use OSC32K which runs at 32.768kHz 
I divide it by 32 (=1,024kHz) in GENDIV and then 1024 with the RTC Prescaler
So the result should be 1Hz.
So I don't understand why there is a difference... I will try with my board tomorrow

thomaseichinger, 2015-01-07:
I know, I reviewed it. :smile: Would be great if we can solve this.

bapclenet, 2015-01-08:
Ca you try to select the 1Hz output on your board by doing that (rtc_init()):
    SYSCTRL->OSC32K.bit.EN1K = 1;
    SYSCTRL->OSC32K.bit.EN32K = 0;
It doesn't change anything on mine. 
If we can select the 1Hz output, we won't need to divide by 32 in GENDIV and maybe it will solve our problem?

thomaseichinger, 2015-01-08:
No, no change for me neither.

The datasheet p.178 15.8.7 states: 
> Bits 22:16 – CALIB[6:0]: Oscillator Calibration These bits control the oscillator calibration.
This value must be written by the user.
Factory calibration values can be loaded from the non-volatile memory. Refer to “NVM Software Calibration Area Mapping” on page 26.

Or maybe change to use the external crystal oscillator `XOSC32K`?

bapclenet, 2015-01-08:
By using XOSC32K, this doesn't change anything.

For calibration, which value do you suggest then? :-)

bapclenet, 2015-01-08:
I tried this in rtc_init():
<code>SYSCTRL->OSC32K.bit.EN32K = 1;</code>
<code>SYSCTRL->OSC32K.bit.CALIB = (*((uint32_t *) SYSCTRL_FUSES_OSC32K_CAL_ADDR) &     SYSCTRL_FUSES_OSC32K_CAL_Msk) >> SYSCTRL_FUSES_OSC32K_CAL_Pos;</code>
<code>SYSCTRL->OSC32K.bit.ENABLE = 1;</code>

But still the problem doesn't change even if the calibration seems good

thomaseichinger, 2015-01-08:
So can you confirm the drift in general?

bapclenet, 2015-01-08:
Yes there is a drift but I can't find out why

bapclenet, 2015-03-26:
I tried again with ATMEL STUDIO's RTC example, it runs 32 faster than it should be. They select the 1KHz from the 32.768kHz Oscillator and set GENDIV to 0 but it still runs at 32.768kHz  instead of 1. I think it's a hardware problem.
By selecting the 1kHz output, we may be more accurate and then, we won't see this drift.

bapclenet, 2015-03-26:
Ok, we can correct the clock with FREQCORR Value, it should work for us since we all have the SAMR21 XPRO board. 

Here is the formula: 
Correction in PPM =  FREQCORR.VALUE / (1024 * 976) * 10^6 PPM

According to http://www.maximintegrated.com/en/design/tools/calculators/product-design/rtc.cfm and the delay of one and half minutes every hour, our board has a deviation of 25000ppm ...
The register FREQCORR.VALUE  is from 1 to 127.

There is something wrong somewhere

bapclenet, 2015-03-26:
Btw, I tried to set CALIB for SYSCTRL->OSC32K.bit.CALIB with p27 datasheet. 

    SYSCTRL->OSC32K.bit.CALIB = (*(uint32_t *)SYSCTRL_FUSES_OSC32K_CAL_ADDR & (SYSCTRL_FUSES_OSC32K_CAL_Msk))>>SYSCTRL_FUSES_OSC32K_CAL_Pos;
    SYSCTRL->OSC32K.bit.CALIB = (*(uint32_t *)SYSCTRL_FUSES_OSC32K_CAL_ADDR >> SYSCTRL_FUSES_OSC32K_CAL_Pos);

Which one of the two lines seems right for you? @thomaseichinger 
For both of them, after 7 min, the difference is -10 so I don't get it.

bapclenet, 2015-04-08:
As mentionned in the datasheet, we should use the XOSC32K instead of the OSC32K because the OSC32K is not accurate.
By configuring the example provided by ATMEL with XOSC32K, I've got an accurate RTC. 
By using XOSC32K in RIOT, it's not accurate then there is a configuration problem in the RTC driver

bapclenet, 2015-04-09:
I've found out! Will do a PR soon

OlegHahm, 2015-04-09:
:raised_hands:

PeterKietzmann, 2015-04-10:
This issue was fixed with #2767

closed 2015-04-10 by PeterKietzmann