RIOT-OS/RIOT #2263: native: weak random generator

d00616, 2015-01-07:
This random generator is not suitable for cryptography. This is an pseudo random number generator without any real random component.

micro-ecc signature tests are failing. It looks like at some point the generated random date are "0"'s or the length of generated data are 0. When i use RNG provided by micro-ecc all tests are working fine. On an NRF51 MCU with integrated random generator all tests are also always correct.

An practibale solution for Windows and Linux can be found here: https://github.com/d00616/micro-ecc/blob/master/uECC.c#L328

LudwigOrtmann, 2015-01-07:
Does it need to be strong?

LudwigOrtmann, 2015-01-07:
It appears to me there are two issues here:

One is weak crypto - which is fine from my point of view, native is not meant for production use. If you think it is not OK for native to have weak crypto please elaborate why.

The other is unclear to me - is it two different effects you are seeing or are is are you uncertain?

d00616, 2015-01-07:
> Does it need to be strong?
Yes. Because a key generated with this RNG depends on initialization of RNG. When i know how the RNG is initialized i can reproduce any key generated with this implementation. On actual native Implementation are only 65533 possible keys because the RNG is initialized with PID.

> One is weak crypto - which is fine from my point of view, native is not meant for production use. 
Today WE know this. But what, if somebody using native implementation to build an base station on Linux ARM hardware? Or somebody is using this code to pregenerate keys for micro controllers without an internal RNG?

I think it's not possible to prevent anybody using this RNG for cryptographical operations in production environments. This can blame RIOT OS.

LudwigOrtmann, 2015-01-07:
The `periph/random` interface documentation specifically advises you to check the quality of the randomness for each board. It is not guaranteed to be strong.

LudwigOrtmann, 2015-01-07:
I am referring to this by the way: http://riot-os.org/api/group__driver__periph__random.html#details

N8Fear, 2015-01-09:
Can't this be easily fixed by using e.g. /dev/random or /dev/urandom (at least on *nix) systems instead of just using the pid?

LudwigOrtmann, 2015-01-10:
Yes, the native RNG could use `/dev/random` but the question is whether this is desirable.

My reasoning so far:
1. The API does not need high quality random numbers. In case an implementation needs it should either choose a platform that has a high-quality RNG or use some other means to produce high-quality random data (a module currently missing in RIOT).
2. `/dev/random` is slow, native's goals include speed.
3. for testing purposes a reproducible behavior is necessary, therefore the current implementation uses a user modifiable seed.

If someone can provide me with a good reason why high quality random data is necessary for their application with native, I'm happy to implement `/dev/random` reading as an *additional* source for random data.

Just for the record: for technical reasons native does not have and won't get support for non-*nix systems.

d00616, 2015-01-10:
At the moment there are two RNG implementations. An pseudo RNG i expect its fast an reproducible and an periph/random implementation i expect it provides best random data available by platform (including native).

What about changing the native RNG using random until "-i" or an additional argument is provided?

LudwigOrtmann, 2015-01-10:
@d00616  https://github.com/RIOT-OS/RIOT/pull/2274 << this implements it.

d00616, 2015-01-10:
@LudwigOrtmann This looks like an very good compromise. Thanks.

LudwigOrtmann, 2015-01-14:
Implemented via #2274

closed 2015-01-14 by LudwigOrtmann