RIOT-OS/RIOT #2269: IPv6: upper layer checksum calculation broken?

OlegHahm, 2015-01-08:
@haukepetersen just made me aware that according to [RFC 2460, 8.1](https://tools.ietf.org/html/rfc2460#section-8.1) the upper layer checksum is calculated over a pseudo header that contains source  **and** destination address.
However, for the calculation in RIOT the source address is used twice while the destination address is completely ignored:
```C
uint16_t ipv6_csum(ipv6_hdr_t *ipv6_header, uint8_t *buf, uint16_t len,
uint8_t proto)
{
uint16_t sum = 0;
DEBUG("Calculate checksum over src: %s, dst: %s, len: %04X, buf: %p, proto: %u\n",
ipv6_addr_to_str(addr_str, IPV6_MAX_ADDR_STR_LEN,
&ipv6_header->srcaddr),
ipv6_addr_to_str(addr_str, IPV6_MAX_ADDR_STR_LEN,
&ipv6_header->destaddr),
len, buf, proto);
sum = len + proto;
sum = net_help_csum(sum, (uint8_t *) &ipv6_header->srcaddr,
2 * sizeof(ipv6_addr_t));
sum = net_help_csum(sum, buf, len);
return (sum == 0) ? 0xffff : HTONS(sum);
}
```
(https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/sixlowpan/ip.c#L888)

Apparently the thing is done in Contiki (https://github.com/contiki-os/contiki/blob/master/core/net/ipv6/uip6.c#L353), but I couldn't find no hint that anyone ever complained about this problem. That makes me wonder if I am missing something here or if we're a facing a bug in both network stacks that was silently ignored for years. If the latter: shouldn't we have seen errors in Wireshark for every UDP transmission then?

authmillenon, 2015-01-08:
I was wondering the same thing when I took this for the new IPv6 layer in the netapi port. I went on then to reimplement it according to RFC, but did not test it with wireshark yet.

> If the latter: shouldn't we have seen errors in Wireshark for every UDP transmission then?

There were some Heisenerrors with ICMP packets.

cgundogan, 2015-01-08:
When I send udp packets from within the rpl_udp example(master), wireshark does not complain about the checksum (I activated the checksum validation in the preferences tab)

OlegHahm, 2015-01-08:
Ah, my fault, @adamdunkels enlightened me: https://github.com/contiki-os/contiki/issues/940#issuecomment-69195400 - sorry for the noise.

closed 2015-01-08 by OlegHahm

cgundogan, 2015-01-08:
the checksum is calculated by the function `ipv6_csum`. There are two calls to `net_help_csum` and each one of this call moves the `buf` variable. The `dest`address follows the `src` address immediatly in the struct. Maybe not the best idea to rely on this, because of unknown paddings and such