RIOT-OS/RIOT #2270: rpl: root-to-node communication in non-storing mode

cgundogan, 2015-01-08:
It seems that the communication from root to node does not happen.
While packets can travel from node to node through the root by adding a source routing header, all packets originating from the root itself bypass RPL (do not enter any RPL relevant code). Thus, no source routing header is added to the packets and they get lost.

authmillenon, 2015-01-08:
I don't know anything about RPL non-storing mode, that's why I reassign.

cgundogan, 2015-01-09:
I see. The problem lies somewhere within the network layer, more precisely - in ip.c .
Therefore, I thought you might be the right person to choose. I will assign this to me then.

My gut tells me it should be an easy fix. When sending out a packet via `ipv6_sendto` or `ipv6_send_packet`, the packet needs to be intercepted by the RPL implementaton first in order to attach the source routing header.

fabianbrandt, 2015-01-13:
Shame on me. The logic for that was there, but somehow not in the final version :(. The fix suggestion is right however.

PeterKietzmann, 2015-03-04:
Is this already fixed somewhere? Or can we reference a PR?

emmanuelsearch, 2015-04-24:
@cgundogan what's the status on this? Can we close this issue?

cgundogan, 2015-08-17:
obsolete, because of the introduction of a new network stack

closed 2015-08-17 by cgundogan