RIOT-OS/RIOT #2276: samd21/periph/spi: spi_transfer_regs counts strangely

LudwigOrtmann, 2015-01-10:
Example: accounting SPI bytes upon packet reception on the `samr21-xpro`:
```
in*: 0x20001cb5 in: 0 transfered: 2
at86rf231_read_fifo: read 2/1 bytes
in*: 0x20001b46 in: 0 transfered: 24
at86rf231_read_fifo: read 36/18 bytes
```
with
```diff
diff --git a/cpu/samd21/periph/spi.c b/cpu/samd21/periph/spi.c
index d921e42..23d219d 100644
--- a/cpu/samd21/periph/spi.c
+++ b/cpu/samd21/periph/spi.c
@@ -217,7 +217,7 @@ int spi_transfer_bytes(spi_t dev, char *out, char *in, unsigned int length)
     int transfered = 0;
 
     if (out != NULL) {
-        DEBUG("out*: %p out: %x length: %x\n", out, *out, length);
+        printf("out*: %p out: %x length: %x\n", out, *out, length);
         while (length--) {
             int ret = spi_transfer_byte(dev, *(out)++, 0);
             if (ret <  0) {
@@ -234,10 +234,10 @@ int spi_transfer_bytes(spi_t dev, char *out, char *in, unsigned int length)
             }
             transfered += ret;
         }
-        DEBUG("in*: %p in: %x transfered: %x\n", in, *(in-transfered), transfered);
+        printf("in*: %p in: %x transfered: %x\n", in, *(in-transfered), transfered);
     }
 
-    DEBUG("sent %x byte(s)\n", transfered);
+    printf("sent %x byte(s)\n", transfered);
     return transfered;
 }
```
and
```diff
diff --git a/drivers/at86rf231/at86rf231_spi.c b/drivers/at86rf231/at86rf231_spi.c
index 72fb0ff..c71c5a3 100644
--- a/drivers/at86rf231/at86rf231_spi.c
+++ b/drivers/at86rf231/at86rf231_spi.c
@@ -53,8 +53,9 @@ void at86rf231_read_fifo(uint8_t *data, radio_packet_length_t length)
     /* Start the SPI transfer */
     gpio_clear(AT86RF231_CS);
     /* Read a number of bytes from the devices frame buffer */
-    spi_transfer_regs(AT86RF231_SPI, AT86RF231_ACCESS_FRAMEBUFFER | AT86RF231_ACCESS_READ,
+    int i = spi_transfer_regs(AT86RF231_SPI, AT86RF231_ACCESS_FRAMEBUFFER | AT86RF231_ACCESS_READ,
                       0, (char*)data, length);
+    printf("at86rf231_read_fifo: read %i/%i bytes\n", i, length);
     /* End the SPI transfer */
     gpio_set(AT86RF231_CS);
 }
```

PeterKietzmann, 2015-03-04:
Too much for a newbie task?

bapclenet, 2015-06-28:
@LudwigOrtmann, is the problem solved or does it still happen?

PeterKietzmann, 2015-06-29:
This is related to the old driver implementation. Would be nice if someone checks if this also occurs in the new implementation somehow. @bapclenet any capacities :-) ?

gebart, 2015-06-29:
There is a PR open for changing the api as well.

bapclenet, 2015-06-29:
@PeterKietzmann, I'm currently on an other project, I will be back with the samr21 later.
Gebart, yes it might solve the problem but I think this is specific to the samr21.

thomaseichinger, 2016-04-03:
Closing as the driver was updated and the described problem should be solved.
@LudwigKnuepfer Please reopen if you disagree.

closed 2016-04-03 by thomaseichinger