RIOT-OS/RIOT #2278: netapi: Port all layers to netapi or create new modules using netapi

authmillenon, 2015-01-12:
Based in part on the very helpful feature list of (CiAO/IP)[http://www.danceos.org/publications/MobiSys-Draft-2012-Borchert.pdf].

This list might even become further sub-devided in the future, if possible:

- [x] Paket buffer (@authmillenon, needs adaption to #2158 though)
- [ ] Centralized Checksum Calculation (Checksums depends on Pseudo header => containing source address is determined in IP layer => upper layer need some kind of callback)
- [ ] IPv6
  - [ ] IPv6 Tx/Rx (@authmillenon, #2003)
  - [ ] ICMPv6 ([RFC 4443](https://tools.ietf.org/html/rfc4443), @Lotterleben for general API)
    - [ ] ICMPv6 Error Messages ***(does not exist in current implementation)***
    - [ ] ICMPv6 Echo Request/Reply (Ping)
    - [ ] IPv6 Extension Headers ***(do not exist as API, just SRH for now)***
    - [ ] NDP ([RFC 4861](https://tools.ietf.org/html/rfc4861))
      - [ ] Neighbor Cache ***(needs some cleanup, currently very 6LoWPAN focused)***
      - [ ] Neighbor Solicitation/Advertisement
        - [ ] Source/Target Link-layer address option
      - [ ] Router Solicitation/Advertisement
        - [ ] Source Link-layer address option
        - [ ] Prefix information option ***(does not exist in current implementation)***
        - [ ] MTU option ***(does not exist in current implementation)***
      - [ ] NDP state machine ***(does not exist as clearly as desired in current implementation)***
      - [ ] Redirect
        - [ ] Redirected Header option ***(does not exist in current implementation)***
- [ ] 6LoWPAN (@authmillenon, #1680)
  - [x] Fragment Reassembly/Fragmentation
  - [ ] Header Compression according to [RFC 6282](https://tools.ietf.org/html/rfc6282)
    - [x] IPHC compression/decompression (as part of #1680)
    - [ ] NHC (next header compression)
      - [ ] IPv6 Extension Headers
      - [ ] UDP
- [ ] IPv4 ***(does not exist currently)***
  - [ ] IPv4 Tx/Rx
  - [ ] IPv4 Fragment Reassembly/Fragmentation
  - [ ] ARP
    - [ ] ARP Reply/Request
    - [ ] ARP Cache ***(maybe merge data base with NDP neighbor cache and Forwarding Table?)***
  - [ ] ICMP
- [ ] Routing
  - [ ] RPL (@OlegHahm, @cgundogan, @fabianbrandt, …)
  - [ ] AODVv2 (@Lotterleben)
  - [ ] OLSR (@benpicco)
  - [ ] Forwarding Table (@bytesgalore, #2211)
- [ ] UDP (@cgundogan?)
  - [ ] UDP Tx/Rx
  - [ ] UDP Checksumming
- [ ] TCP (there was a BSc. student interested)
  - [ ] TCP client (`connect()`) / server (`listen()`)
  - [ ] Sliding Window Tx/Rx
  - [ ] Avoid Silly Window Syndrom Tx/Rx
  - [ ] RTT estimation
  - [ ] Congestion Control (Slow-Start)
  - [ ] TCP urgent Data
  - [ ] LImit EXcessive Retransmissions
  - [ ] MSS Option
  - [ ] Timestamp Option
  - [ ] Multipath-TCP

authmillenon, 2015-02-08:
@OlegHahm can we please use the Network Stack Task Force Milestone only for PR's that are explicitly mentioned in this list? Otherwise we risk getting people easily confused.

authmillenon, 2015-02-19:
The last critical component was merged. From now on it's Happy Hacking!

OlegHahm, 2015-02-19:
> @OlegHahm can we please use the Network Stack Task Force Milestone only for PR's that are explicitly mentioned in this list? Otherwise we risk getting people easily confused.

Ah, haven't seen this before. Will do.

authmillenon, 2015-03-07:
> @OlegHahm can we please use the Network Stack Task Force Milestone only for PR's that are explicitly mentioned in this list? Otherwise we risk getting people easily confused.

Just a little update on my rational for assigning the milestone: PR's or features that are explicitly mentioned here and any fix- or follow-up to those.

authmillenon, 2015-09-23:
Updated list.

miri64, 2016-10-17:
Though there are still some things under construction, since PRs exist I would deem this issue as complete, so close.

closed 2016-10-17 by miri64