RIOT-OS/RIOT #2281: travis: add check for line endings

OlegHahm, 2015-01-12:
Add a script that check for valid UNIX line endings.

kushalsingh007, 2015-03-04:
I will be working on this issue , before I start just a question , once the script is run what should be the expected output format (should there be a list of the places where UNIX line endings are not used) ?
Example format--
error: Filename: Line number

kushalsingh007, 2015-03-05:
Do tell if the following code looks good , tell the changes that need to be done (if any) so that I can create a PR for it. ( Note : I removed the initial copyright part from now , I think there wont be any changes for that part )

Here is the link to the code I have written  

https://gist.github.com/kushalsingh007/2cdbc043b649550c8420

gebart, 2015-03-05:
@kushalsingh007 I think printing the name of the offending file is enough to know what to do. Most likely the entire file is Windows line endings and then it won't be much help to print every line number.

gebart, 2015-03-05:
I am sorry, but why do we need to implement anything at all for this? egrep handles it just fine:

    egrep '^M$' . -R -l

where `^M` is the character that comes up when you press `ctrl+v`, `enter` in the terminal.

kushalsingh007, 2015-03-05:
So @gebart should I then create a bash script which uses egrep ? I guess I should waited for someone before starting the task to elaborate the comment :( . Regardless , the script also works fine doesn't it ?

gebart, 2015-03-05:
@kushalsingh007 that's just my opinion. Since this script will have to be run for every PR and every commit made to the RIOT repo on the overloaded Travis CI servers, it is better to use the option which has the least overhead, which I guess is egrep or some other tool which has been around for years instead of writing our own.

Writing this kind of tool is a good student assignment but I think for everyday usage in our CI it is better to go with something tried and tested.

Assigned @OlegHahm since @Kijewski has not been very active on RIOT lately.

kushalsingh007, 2015-03-05:
@gebart  Here is the updated python script regardless https://gist.github.com/kushalsingh007/2cdbc043b649550c8420
I think what you say sounds correct :)
 , should I write a bash script using egrep now , or wait till @OlegHahm comes ?

OlegHahm, 2015-03-05:
Yes, I agree with @gebart that egrep aka `grep -E` seems to be the best solution

kushalsingh007, 2015-03-05:
Have a look at the bash script and give your feedback 
 https://gist.github.com/kushalsingh007/05e50ade69ea3c98ee6f

kushalsingh007, 2015-03-10:
I think this issue can be closed now.

authmillenon, 2015-03-10:
Yes. One Github-Tip: If you reference an issue to be fixed in an commit or pull request like "Fixes #<issue no>" or "Closes #<issue no>" the issue gets to be closed automatically as soon as it gets merged ;-)

closed 2015-03-10 by authmillenon