RIOT-OS/RIOT #2284: USB Support

Farthen, 2015-01-12:
It could be desirable to support USB and USB OTG which is present on many devices currently running RIOT.

This could have multiple use cases depending on the device support:
* USB serial console
* Sending network and/or other data from/to a computer running control or monitoring software.
* Updating application code (it was discussed on the ML a few days ago to have relocatable? applications that can be loaded in a running RIOT system, circumventing license issues and allowing for easier application updates; is there an issue for this?)

I would like to use this issue to start designing a platform independent API for USB support if other people are interested in having such a feature.

gebart, 2015-01-12:
I am interested in such a feature as well. Both for serial console as well as a network device.

Troels51, 2015-01-29:
I've been looking into this as well, knowing nothing about USB i looked for APIs in other systems, here are  the ones i found.
* [ARM CMSIS DRIVER](http://www.keil.com/pack/doc/CMSIS/Driver/html/group__usb__interface__gr.html)
* [TI MSP USB DEV](http://www.ti.com/tool/msp430usbdevpack)
They could be used as inspiration

jfischer-phytec-iot, 2015-01-29:
@gebart I am on it for kinetis_common, for the beginning bulk transfers only.

gebart, 2015-01-29:
@jfischer-phytec-iot nice! Looking forward to this. I can assist with testing when you have something you'd like to share. Unfortunately, I do not have time to work on developing this feature for the time being.

haukepetersen, 2015-01-30:
+1 for USB support! I always wanted to connect old USB-mice or gamepads to micro-controllers to play around with... :-)

OlegHahm, 2015-01-30:
And keyboard of course.

OlegHahm, 2015-01-30:
What about VGA? ;)

jfischer-phytec-iot, 2015-01-30:
but of course via "Cup Heater + 4 Port USB Hub"

drasko, 2015-09-15:
Any news on this?

OlegHahm, 2015-09-16:
Please see https://github.com/RIOT-OS/RIOT/issues/3546

PeterKietzmann, 2019-07-23:
With #10916 and following PRs merged, I think we can close this issue now.

closed 2019-07-23 by PeterKietzmann