RIOT-OS/RIOT #2291: sys/net/include: move to /sys/include/net?

haukepetersen, 2015-01-13:
Just thinking out loud: does it make sense to move the network stack headers to `sys/include/net` and reference them in the code like `#include "net/ipv6.h"`?

This has in my opinion two advantages:

1. network layer includes are clearly distinguishable
2. and we would save about ~35 lines stating `*INCLUDE += sys/net/include`

gebart, 2015-01-13:
:+1: it makes sense to reduce the number of individual `-I{DIR}` include directories. 
Another bonus when writing `#include "net/ipv6.h"` is that it shows clearly which component the included header is supposed to belong to (ipv6.h is quite obvious but there may be others with less obvious names), makes it easier to find the right file while developing without some heavy IDE like Eclipse.

OlegHahm, 2015-01-13:
Sound very reasonable to me, too.

LudwigOrtmann, 2015-01-13:
I'm not opposed.

authmillenon, 2015-01-13:
I'm not opposed, also.

authmillenon, 2015-01-13:
Though I'm not sure if the `net/` prefix is necessary… I'm thinking of modules that might have submodules: e.g. `net/sixlowpan/hc.h` vs. `sixlowpan/hc.h`

haukepetersen, 2015-01-13:
I think it does not hurt. This way the `net/` prefix labels consistently everything belonging to the network stack and its only 4 chars...

PeterKietzmann, 2015-03-04:
That already happened for the new network stuff, right? Is this issue "closable"?

authmillenon, 2015-03-04:
Yes on new stuff, but No on "closable". There is (and will be until complete deployment of `ng_`) still stuff in `sys/net/include`.

haukepetersen, 2015-03-05:
Actually I say we can close the issue, as the decision for `sys/include/net/...` was made...

PeterKietzmann, 2015-03-05:
Ok that is what was in my mind. @authmillenon should do this (at will)

authmillenon, 2015-03-05:
Fine…

closed 2015-03-05 by authmillenon