RIOT-OS/RIOT #2298: timex_t functions byReference instead of byValue?

cgundogan, 2015-01-13:
Sometimes I get the warning/error - not sure what it is - from travis/cppcheck saying:
```
sys/include/timex.h:70: performance (passedByValue): Function parameter 'a' should be passed by reference.
sys/include/timex.h:70: performance (passedByValue): Function parameter 'b' should be passed by reference.
sys/include/timex.h:80: performance (passedByValue): Function parameter 'a' should be passed by reference.
sys/include/timex.h:80: performance (passedByValue): Function parameter 'b' should be passed by reference.
sys/include/timex.h:102: performance (passedByValue): Function parameter 'a' should be passed by reference.
sys/include/timex.h:102: performance (passedByValue): Function parameter 'b' should be passed by reference.
sys/include/timex.h:135: performance (passedByValue): Function parameter 'a' should be passed by reference.
```
[#2155#issuecomment-69809181](https://github.com/RIOT-OS/RIOT/pull/2155#issuecomment-69809181)

Looking at the function definitions:
```
timex_t timex_add(const timex_t a, const timex_t b);
timex_t timex_sub(const timex_t a, const timex_t b);
...
```
Wouldn't it make sense to really change these parameters to be pointers instead?
What's you view on this?

Kijewski, 2015-01-13:
```c
#include <stdint.h>

typedef struct {
    uint32_t seconds;
    uint32_t microseconds;
} timex_t;

extern timex_t add1(const timex_t a, const timex_t b);
extern timex_t add2(const timex_t *a, const timex_t *b);
extern void add3(const timex_t *a, const timex_t *b, timex_t *r);

extern timex_t a, b, r;

void test_add1(void) {
    r = add1(a, b);
}

void test_add2(void) {
    r = add2(&a, &b);
}

void test_add3(void) {
    add3(&a, &b, &r);
}
```

Interesting are the push and pop operations:

i386:
```
00000000 <test_add1>:
   0:   83 ec 1c                sub    $0x1c,%esp
   3:   8d 44 24 08             lea    0x8(%esp),%eax
   7:   83 ec 0c                sub    $0xc,%esp
   a:   ff 35 04 00 00 00       pushl  0x4
  10:   ff 35 00 00 00 00       pushl  0x0
  16:   ff 35 04 00 00 00       pushl  0x4
  1c:   ff 35 00 00 00 00       pushl  0x0
  22:   50                      push   %eax
  23:   e8 fc ff ff ff          call   24 <test_add1+0x24>
  28:   8b 44 24 24             mov    0x24(%esp),%eax
  2c:   8b 54 24 28             mov    0x28(%esp),%edx
  30:   a3 00 00 00 00          mov    %eax,0x0
  35:   89 15 04 00 00 00       mov    %edx,0x4
  3b:   83 c4 38                add    $0x38,%esp
  3e:   c3                      ret    

00000000 <test_add2>:
   0:   83 ec 1c                sub    $0x1c,%esp
   3:   8d 44 24 08             lea    0x8(%esp),%eax
   7:   52                      push   %edx
   8:   68 00 00 00 00          push   $0x0
   d:   68 00 00 00 00          push   $0x0
  12:   50                      push   %eax
  13:   e8 fc ff ff ff          call   14 <test_add2+0x14>
  18:   8b 44 24 14             mov    0x14(%esp),%eax
  1c:   8b 54 24 18             mov    0x18(%esp),%edx
  20:   a3 00 00 00 00          mov    %eax,0x0
  25:   89 15 04 00 00 00       mov    %edx,0x4
  2b:   83 c4 28                add    $0x28,%esp
  2e:   c3                      ret    

00000000 <test_add3>:
   0:   83 ec 10                sub    $0x10,%esp
   3:   68 00 00 00 00          push   $0x0
   8:   68 00 00 00 00          push   $0x0
   d:   68 00 00 00 00          push   $0x0
  12:   e8 fc ff ff ff          call   13 <test_add3+0x13>
  17:   83 c4 1c                add    $0x1c,%esp
[... 136 lines cut ...]
  26:   82 4c 00 00     mov     r12,    &0x0000
  2a:   82 4d 00 00     mov     r13,    &0x0000
  2e:   82 4e 00 00     mov     r14,    &0x0000
  32:   82 4f 00 00     mov     r15,    &0x0000
  36:   30 41           ret

00000000 <test_add2>:
   0:   3e 40 00 00     mov     #0,     r14     ;#0x0000
   4:   3f 40 00 00     mov     #0,     r15     ;#0x0000
   8:   b0 12 00 00     call    #0x0000
   c:   82 4c 00 00     mov     r12,    &0x0000
  10:   82 4d 00 00     mov     r13,    &0x0000
  14:   82 4e 00 00     mov     r14,    &0x0000
  18:   82 4f 00 00     mov     r15,    &0x0000
  1c:   30 41           ret

00000000 <test_add3>:
   0:   3d 40 00 00     mov     #0,     r13     ;#0x0000
   4:   3e 40 00 00     mov     #0,     r14     ;#0x0000
   8:   3f 40 00 00     mov     #0,     r15     ;#0x0000
   c:   b0 12 00 00     call    #0x0000
  10:   30 41           ret
```

In terms of stack space I think we have a winner.

OlegHahm, 2015-01-14:
There was some discussion already here: https://github.com/RIOT-OS/RIOT/pull/1843#issuecomment-60270374
I once started to change all these functions but got tired of the cumbersome work.

LudwigOrtmann, 2015-01-14:
I changed the classification of this issue to reflect the answer to the former question.

kaspar030, 2015-01-14:
@Kijewski Nice comparison. Makes me want a script that creates it...

kaspar030, 2015-01-17:
@Kijewski Could you compare again compiling with LTO?

Kijewski, 2015-01-17:
This is the line that I used to get the data:
```sh
msp430-gcc -fdata-sections -ffunction-sections -fomit-frame-pointer -ggdb3 -Os -c x.c && msp430-objdump -dC x.o
```
Seeing LTO in action will be more difficult to test.

closed 2019-09-10 by stale[bot]