RIOT-OS/RIOT #2300: qemu unittest fails because of a page fault

LudwigOrtmann, 2015-01-14:
```
Starting QEMU: qemu-system-i386 -serial tcp:0.0.0.0:35873 -nographic -monitor /dev/null -kernel /home/travis/build/RIOT-OS/RIOT/tests/unittests/bin/qemu-i386/unittests.hex -m 512m
2015-01-14 08:51:56.967893: 2015-01-14 08:51:56.968187: Interrupt handling initialized
PIC initialized
2015-01-14 08:51:56.968621: Warning: Your hardware does not support the NX bit!
2015-01-14 08:51:56.968868: Setting PT_XD = 0.
2015-01-14 08:51:56.969707: Kernel memory: 0x100000 - 0x16e000
2015-01-14 08:51:56.969984: .text: 0x100050 - 0x137c30
2015-01-14 08:51:56.970216: .rodata: 0x138000 - 0x144144
2015-01-14 08:51:56.970498: .data: 0x14f000 - 0x1505dc
2015-01-14 08:51:56.970735: .bss: 0x151000 - 0x16db34
2015-01-14 08:51:56.971000: Unmapped memory: 0x16e000 - 0x170000
2015-01-14 08:51:56.971169: Heap start: 0x170000
2015-01-14 08:51:56.971543: 0000000000000000 - 000000000009fc00 (usable)
2015-01-14 08:51:56.971942: 000000000009fc00 - 00000000000a0000 (unusable: reseved)
2015-01-14 08:51:56.972340: 00000000000f0000 - 0000000000100000 (unusable: reseved)
2015-01-14 08:51:56.972686: 0000000000100000 - 000000001fffe000 (usable)
2015-01-14 08:51:57.235538: 000000001fffe000 - 0000000020000000 (unusable: reseved)
2015-01-14 08:51:57.236000: 00000000fffc0000 - 0000000100000000 (unusable: reseved)
2015-01-14 08:51:57.238810: There are 130704 free pages (510.562 MB) available for the heap.
2015-01-14 08:51:57.239045: Virtual memory initialized
2015-01-14 08:51:57.239522: RTC initialized [08:51:56, 2015-01-14]
2015-01-14 08:51:57.239683: PIT initialized
2015-01-14 08:51:57.239854: Measuring CPU speed:
2015-01-14 08:51:57.427047: CPU speed = 2195.228 MHz
2015-01-14 08:51:58.001082: 08:51:57, 2015-01-14 is 2937168881
2015-01-14 08:51:58.001477: Looking up PCI devices
2015-01-14 08:51:58.002487: 00:00.0 "Intel Corporation": "440FX - 82441FX PMC [Natoma]" (Bridge device: Host bridge, rev: 02)
2015-01-14 08:51:58.003325: 00:01.0 "Intel Corporation": "82371SB PIIX3 ISA [Natoma/Triton II]" (Bridge device: ISA bridge, rev: 00)
2015-01-14 08:51:58.004337: 00:01.1 "Intel Corporation": "82371SB PIIX3 IDE [Natoma/Triton II]" (Mass storage controller: IDE controller, rev: 00)
2015-01-14 08:51:58.004765: BAR 4: I/O space, ports 0xc040-0xc04f
2015-01-14 08:51:58.005546: 00:01.3 "Intel Corporation": "82371AB/EB/MB PIIX4 ACPI" (Bridge device: Other bridge device, rev: 03)
2015-01-14 08:51:58.005820: IRQ: new = 10, old = 9
2015-01-14 08:51:58.006573: 00:02.0 "Cirrus Logic": "GD 5446" (Display controller: VGA-compatible controller, rev: 00)
2015-01-14 08:51:58.023327: Page fault: access violation while writing to non-present page.
2015-01-14 08:51:58.023735: EAX=21fffff8 ECX=00000001 EDX=2000fffc EBX=00000000
2015-01-14 08:51:58.024132: ESP=00006c30 EBP=00006c50 ESI=00000000 EDI=0226a000
2015-01-14 08:51:58.024289: Error code=00000002
2015-01-14 08:51:58.024664: CR0=80010031 CR2=21fffff8 CR3=00158000 CR4=000001e0
2015-01-14 08:51:58.024907: EIP=0008:00107a60 EFLAGS=00000016
2015-01-14 08:51:58.025023: <stack trace>
2015-01-14 08:51:58.025122: 00107a60
2015-01-14 08:51:58.025190: ???
2015-01-14 08:51:58.025311: </stack trace>
2015-01-14 08:51:58.025411: Halting.
```

gebart, 2015-01-14:
Why do these seemingly random Travis failures keep popping up? Are they running out of memory or something?

LudwigOrtmann, 2015-01-14:
I've never seen this kind of failure before. I was under the impression that this is not a random failure.

gebart, 2015-01-14:
Okay, I did not really look into the problems before, I thought this was the same as all the comments in almost every PR "Kicked Travis"

LudwigOrtmann, 2015-01-14:
The random fails should be discussed in https://github.com/RIOT-OS/RIOT/issues/1377

miri64, 2016-10-17:
Due to the inclusion of `relic` and since `periph_timer` is now required unittests don't build for this board anymore in any case :(.

PeterKietzmann, 2017-11-27:
qemu support hast been removed with `#8042`

closed 2017-11-27 by PeterKietzmann