RIOT-OS/RIOT #2302: Synchronization: Atomic counters

gebart, 2015-01-14:
We currently have `atomic_set_return()` for atomic variables used by e.g. mutexes.
This works well for a simple on/off mutex lock, but not for counter variables, since the value to set must be computed before calling `atomic_set_return()`.

What are your opinions on adding this kind of functionality to the core atomic API?

Below is an example of an implementation of an increment/decrement function for an atomic counter without disabling interrupts (for a Cortex-M CPU):

    static inline void exclusive_increment(volatile uint32_t* value) {
      int status;
      int tmp;
      do {
        /* Load exclusive */
        tmp = __LDREXW(value);

        /* increment counter */
        ++tmp;

        /* Try to write the new value */
        status = __STREXW(tmp, value);
      } while (status != 0); /* retry until load-store cycle was exclusive. */
    }

The STREX instruction returns non-zero if an interrupt (or context switch) occurred between the LDREX and the STREX instructions.

I guess the simplest initial implementation for any other CPUs would be to disable interrupts beforehand and enable them afterwards.

OlegHahm, 2015-01-14:
Funny, I had a similar discussion with @haukepetersen yesterday offline. I think adding a function like this to the core API makes a lot of sense. Alternatively there seems to be a builtin gcc function.

haukepetersen, 2015-01-14:
+1

how about adding 
```c
void atomic_inc(int *val);
void atomic_dec(int *val);
```
to the cores atomic interface? And while we are at it: should we consider renaming all the calls to `atomic_` functions to `atomic_arch_`, as this would mean we could throw out `core/include/atomic.h` in favor of `core/include/arch/atomic_ach.h` and get rid of the re-labeling in `atomic_arch.h`?

gebart, 2015-01-14:
@OlegHahm Which is the builtin function? Does it exist in Clang as well as gcc? Any other compilers?

In the Cortex-M specific case it seems the __STREXW macro expands to __strex() which is not defined within the RIOT tree, I guess it's a builtin in arm-none-eabi-gcc et.al.

gebart, 2015-01-14:
@haukepetersen I would probably wish to have the function return the new or old value (à la `i++` or `++i`)

```c
int atomic_inc(int *val);
int atomic_dec(int *val);
```

haukepetersen, 2015-01-14:
oh yes, makes sense.

haukepetersen, 2015-01-14:
what about the type, you think `int` is fine, or should we think about using `unsigned` values as in `atomic_set_return`?

gebart, 2015-01-14:
I think signed makes sense in this case.
Would it make sense to make several variations? int8_t, int16_t etc?

haukepetersen, 2015-01-14:
multiple variations does not make sense in my opinion. I would completely go against the core's design objective as keeping it as minimal as possible...

gebart, 2015-01-14:
@haukepetersen `int` is the most useful type in my opinion.

OlegHahm, 2015-01-14:
I would vote for signed int, too.

OlegHahm, 2015-01-14:
@gebart, concerning the builtins a very brief search yesterday revealed this posting:https://gcc.gnu.org/ml/gcc-help/2014-07/msg00115.html
I haven't checked more in detail yet.

gebart, 2015-01-14:
The implementation I posted in the initial post is inspired by Joseph Yiu "The Definitive Guide to ARM® Cortex®-M3 and Cortex®-M4 Processors"

https://books.google.se/books?id=9YxqAAAAQBAJ&pg=PA353&lpg=PA353&dq=joseph+yiu+exclusive+counter&source=bl&ots=Ua42H7bAez&sig=HjapDL3t5rNaH-3un0wJLuPrPg4&hl=en&sa=X&ei=FFG2VPeHDaXOyQP-xYHwAw&ved=0CC4Q6AEwAw#v=snippet&q=__ldrexw&f=false

Kijewski, 2015-01-16:
Wouldn't it be the easiest solution to only have [compare and swap](http://en.wikipedia.org/wiki/Compare-and-swap)?

```c
extern int atomic_cas(volatile int *dest, int old_value, int new_value);

static inline void atomit_incr(volatile int *var) {
  int current;
  do current = *var; while (!atomic_cas(var, current, current + 1));
}

static inline void atomit_decr(volatile int *var) {
  int current;
  do current = *var; while (!atomic_cas(var, current, current - 1));
}

static inline int atomit_set(volatile int *var) {
  return atomic_cas(var, 0, 1);
}

static inline int atomit_reset(volatile int *var) {
  return atomic_cas(var, 1, 0);
}
```

gebart, 2015-01-16:
@Kijewski I guess that would be the preferred solution on CPUs such as x86 with an CMPXCHG instruction. On Cortex-M I think `atomic_cas` would still be implemented as `LDREX`/`STREX`?

Can we still keep all of the `atomic_xxx` functions in `atomic_arch.c` per CPU?

gebart, 2015-01-18:
@Kijewski Are you against introducing `atomic_inc`,  `atomic_dec` as separate functions?

Kijewski, 2015-01-22:
> Are you against introducing atomic_inc, atomic_dec as separate functions?

No, I'm not. I assumed that compare-and-swap was easier to implement on most platforms.

gebart, 2015-05-22:
closed via #2321

closed 2015-05-22 by gebart