RIOT-OS/RIOT #2311: net: passing of addresses through the network stack

haukepetersen, 2015-01-15:
Right now the interfaces for `netapi` and `netdev` seem to complicated to me. To clean them up I talked with @OlegHahm and we came up with the following concept for passing data and headers through the network stack utilizing the pktbuf:

When passing data up the network stack, we always pass the header of the current module encapsulating the header for the target module. For example:
` an ieee802.15.4 network interface receives data. It parses the 802.15.4 header and puts it into the packet buffer, creating a pkt_t structure and marking it as type PROTO_IEEE802154. It also puts the payload into the packet buffer, linking letting the first pkt_t structure point to the payload. Now the network driver and MAC layer just pass the pointer to the 802.15.4 header to the next network stack module (in this case this would be 6LoWPAN...).`
This way, the receiving network module can always decide if it needs to extract data from the previous header (as e.g. 6LoWPAN might be interested in the PAN id). Otherwise it just throws the previous header away.

When passing data down the network stack, a network layer always allocates the header of the targeted network module before passing the data on. For example:
`when sending an UDP packet, the UPD module must allocate the IPV6 header before passing the data to the IP module. This is no problem, since UDP must know something about IP anyway, e.g. for calculating it's checksum...`

Open Questions:
- It might be difficult to know the correct header/frame format when passing data to different link-layers. My proposed solution is, that if the correct format is unknown for whatever reason, the sending module (i.e. IP) can just prepend a dummy header and the receiving module (i.e. the link layer) can just throw this dummy header away and prepend a more fitting header.

What do you think?

First approximations on this can be seen in #2310 and #2297...

authmillenon, 2015-01-16:
There are several reasons why I'm against it (and why I introduced that the addresses are passed as an extra parameter to the adjacent layers [funny enough I came to this conclusion too after a discussion with @OlegHahm =D):

* every header must be parsed or build double in different places, that are mostly not concerned (the layer concerned to work with it, the layer above to get to the address, or vice versa with building) 
* this leads to the problem, that every layer has to include at least the frame structures and parsing capabilities for *every* adjacent layer, which just doesn't scale (see `transceiver` module if you need an example) 
* parsing outside the module concerned breaks the coherence of modules and is just bad design IMHO
* most protocols are designed that way that they do not need anything more then the addresses from the layer below. The only layer where this might not apply is the MAC layer (see the various IEEE 802.15.4 frames + IEEE 802.15.4, but I think that can be done with the `pkt_proto_t` or if nothing else helps expanding netdev
* if I think about it more there will come more reasons to mind =) 

Concerning the PAN ID: for the current version of IPHC it is not needed anymore (it was needed for the now deprecated HC1) so this corner case is not valid anymore. Even if it would be needed it can be passed as part of the address. 

An alternative I thought about to at least simplify the API was to put the (addr_data, addr_len) tuple into pkt_t, but this would result into more Metadata getting stored in the pktbuf, though there is no need for this.

OlegHahm, 2015-01-16:
I'm not sure about the downward direction (from upper layer down the stack), but I have a gut feeling about the other direction:
I agree that it might be conceptional not the cleanest approach, but my intuition tells me that there might be cases where an upper layer might require more than just the addresses. (Even though I cannot provide a concrete example right now.) However, what makes passing lower layer address to the upper layer a cleaner design than passing the full header?

I also don't see the problem of doubled parsing or building. The upper layer should and could (in most cases) just ignore the passed lower layer header if it doesn't need. But passing the full header instead of just a selected subset (i.e. addresses) the design would be much more flexible to all thinkable and unthinkable protocol stacks.

authmillenon, 2015-01-16:
> I agree that it might be conceptional not the cleanest approach, but my intuition tells me that there might be cases where an upper layer might require more than just the addresses.

For these corner-case an extended version of netdev or netapi can always be provided (maybe we need a somewhat cleaner version of netdev.802154 I want to say).

> However, what makes passing lower layer address to the upper layer a cleaner design than passing the full header?

and

> I also don't see the problem of doubled parsing or building. 

Again: The upper layer can be agnostic about the workings in the lower layer and can decide upon the address if what he got makes sense. Example with just address passing:

* MAC packet with long (64-bit) address
* checks if packet is well formatted etc. and extracts what it needs (addresses and payload amongs others) from MAC header
* hands addresses and payload data over to IPv6
* IPv6 checks if address is either a MAC address (6 byte) or an EUI-64 (8 byte) long => success
* IPv6 continues work

Example with passing headers:
* MAC got packet with long (64-bit) address
* checks if packet is well formatted etc. and extracts what it needs from MAC header
* hands raw packet over to IPv6
* IPv6 needs to know how the particular packet is formatted and find out where the address and payload are stored (aka long #ifdef of doom) and extracts them
* IPv6 checks if address is either a MAC address (6 byte) or an EUI-64 (8 byte) long => success
* IPv6 continues work

> The upper layer should and could (in most cases) just ignore the passed lower layer header if it doesn't need. 

How should it ignore it if it has to extract the addresses + payload first?

> But passing the full header instead of just a selected subset (i.e. addresses) the design would be much more flexible to all thinkable and unthinkable protocol stacks.

Again: There I think we should reduce complexity of the API and put flexibility to some corner cases to inheritance.

OlegHahm, 2015-01-16:
> How should it ignore it if it has to extract the addresses + payload first?

It has to know something about the link layer anyway if wants to extract information from the link layer addresses. The case you described will work for 6LoWPAN aka RFC 4944 aka IPv6 over 802.15.4. An implementation of this knows per se about the link layer and could parse the link layer header as simple as it could distinguish between different address types. If it is really agnostic to the link layer, it shouldn't bother about its addresses (and would also not be able to handle them).

cgundogan, 2015-01-16:
Just my two cents..
>OlegHahm - 
>I agree that it might be conceptional not the cleanest approach, but my intuition tells me that there might be cases where an upper layer might require more than just the addresses. (Even though I cannot provide a concrete example right now.) 

I can think of routing protocols trying to benefit from link-centric information to improve their paths or to find broken paths faster (without waiting a full timeout). There are even optimized TCP implementations using more information from lower layers.

I would like to see the possibility to obtain such information from lower and maybe upper layers.

authmillenon, 2015-01-16:
@OlegHahm wrote:
> It has to know something about the link layer anyway if wants to extract information from the link layer addresses. The case you described will work for 6LoWPAN aka RFC 4944 aka IPv6 over 802.15.4. [...]

With the current API it does not. And please remember that we are not only talking about the commutication of Link-layer and 6LoWPAN, but also about the communication of Link-layer and Network-layer, Network-layer and Transport-layer, and all thinkable intermediate layers. 6LoWPAN just needs the address of the 802.15.4 frame. I already implemented this, it worked out quiet well! ;-) True RFC 4944 states you need the PAN-ID for HC1, but you can either argue that it is

a) just part of the address as @rousselk argued when he introduced the predecessor of [`netdev_802154_node_addr_t`](https://github.com/RIOT-OS/RIOT/blob/d4166facccb01d89e239a9211c61e1d5c2ce6f29/drivers/include/netdev/802154.h#L128)
b) or - what I did - say we don't need HC1 (since we have already IPHC as defined in RFC 6282), thus do not need the PAN-ID

@cgundogan wrote:
> I would like to see the possibility to obtain such information from lower and maybe upper layers.

We are talking here about header information of more than two layers if I see this correctly. That would be a completely new addition to the network stack API, IMHO.

authmillenon, 2015-01-16:
Okay alternative: let's have a look at my alternative to #2285: https://github.com/authmillenon/RIOT/commit/9b6c9f3334c98cb84c7c814b875d882e67444927. This would allow for either moving of both the `payload_data` part of a packet and (with a little bit of tweeking) inclusion of header information of lower layers. This would like on receive in an IEEE802.15.4 device e.g. (mind you with #2285 this is way more complicated, since):

```C
pkt_add_header(pkt, pkt->payload_data, sizeof(ieee802154_frame_t));
pkt->headers->header_proto = PKT_PROTO_IEEE802154;
pkt->payload_data += sizeof(ieee802154_frame_t);
ieee802154_frame_t frame = pkt->headers->header_data;
receive_callback(dev, frame->src, frame->src_len, frame->dest, frame->dest_len, pkt);  // (yes I know you can't read the address lengths like that, this is a scetch ;-))
``` 

This would be a good compromise I guess: getting the address data (which most layers definitely need) and also providing the header data, in case some layer needs it later. What do you think?

authmillenon, 2015-01-16:
For a higher layer this would be a `NETAPI_RCV_CMD` IPC call, of course ;-)

authmillenon, 2015-01-16:
with a small adaptation to #2310 `receive_callback()` would be replaced e.g. with

```C
netapi_pkt_t rcv;
rcv.src = frame->src;
rcv.src_len = (frame->fcf->src_mode == IEEE802154_SRC_MODE_SHORT) ? 2 : 8;
rcv.dest = frame->dest;
rcv.dest_len = (frame->fcf->dest_mode == IEEE802154_SRC_MODE_SHORT) ? 2 : 8;
rcv.pkt = pkt;
event_callback(dev, NETDEV_EVENT_RX_COMPLETE, &rcv);

OlegHahm, 2015-01-16:
> With the current API it does not.

Of course, it does how would it be possible to make this distinction:
> IPv6 checks if address is either a MAC address (6 byte) or an EUI-64 (8 byte) long => success
?

> And please remember that we are not only talking about the commutication of Link-layer and 6LoWPAN, but also about the communication of Link-layer and Network-layer, Network-layer and Transport-layer, and all thinkable intermediate layers. 

Which makes it even more attractive to me. I haven't checked (and am not a transport layer expert), but maybe there are transport layer protocols that uses certain information from IP headers.

> 6LoWPAN just needs the address of the 802.15.4 frame. I implemented this, it worked out quiet well!

Well, this is true for IP over 802.15.4, but maybe upcoming 6lowpan specifications will need other information from the link layer.

authmillenon, 2015-01-16:
>> With the current API it does not.
>
> Of course, it does how would it be possible to make this distinction:

It does not need to (I should have rephrased that). If the address is not as expected the packet is discarded. I always said (maybe not clear enough): The adjacent layers are agnostic to the concerning layers data *except* the for the addresses it uses to address the packets (destination address for sending, source and destination address for receiving to e.g. resolve header compression).

> Which makes it even more attractive to me. I haven't checked (and am not a transport layer expert), but maybe there are transport layer protocols that uses certain information from IP headers.

With my compromise this data is provided, but because of the pointers to the addresses and there length given, the Transport layers that *don't* need certain information from the IP headers do not need to now if they have an IPv4 address or IPv6 address or AppleTalk address and do not need to know anything of the network protocol, except if they can handle addresses of the format given.

> Well, this is true for IP over 802.15.4, but maybe upcoming 6lowpan specifications will need other information from the link layer.

Again: read my comprimise: header information is included ;-)

haukepetersen, 2015-01-17:
After looking through this discussion: Can anyone answer this question: Does IPv4 or IPv6 need to know anything from the Link-Layer when receiving data?

authmillenon, 2015-01-17:
Yes, the Link-layer address for ARP or NDP, respectively. Or did you mean more than that? At least for IPv6 I can say: no. For IPv4 I have not enough expertise.

cgundogan, 2015-01-20:
@haukepetersen I can also think of *Link Layer Feedback* for routing and transport layer protocols, to quickly detect link breakages.

haukepetersen, 2015-01-21:
@cgundogan: thanks for the info, thats what I was looking for.

authmillenon, 2015-05-26:
#2706 fixed this

closed 2015-05-26 by authmillenon