RIOT-OS/RIOT #2314: drivers/periph: making bus drivers thread safe

haukepetersen, 2015-01-15:
As pointed out in #2289, our current bus driver implementations are not thread save. A solution is proposed in #2290 (for the STM32F4 cpu). If we use that proposed solution or if we do something else - either way we need to do this for all cpus and bus drivers. This issue is for tracking the implementation status of this effort.

Thread safe bus drivers:
 - [ ] cc2538
    - [ ] i2c
    - [ ] spi
 - [ ] sam3x8e
    - [ ] i2c
    - [ ] spi
 - [ ] samd21
    - [ ] i2c
    - [ ] spi
 - [ ] stm32f0
    - [ ] i2c
    - [ ] spi
 - [ ] stm32f1
    - [ ] i2c
    - [ ] spi
 - [ ] stm32f3
    - [ ] i2c
    - [ ] spi
 - [ ] stm32f4
    - [ ] i2c
    - [ ] spi -> (in progress #2290)

Question: do we need to make the UART thread safe?

thomaseichinger, 2015-01-15:
> Question: do we need to make the UART thread safe?

If we are at it, let's do this.

thomaseichinger, 2015-01-16:
> Question: do we need to make the UART thread safe?

After thinking about it some more it would break usage of DEBUG() in ISR context and similar, so probably not. Other opinions?=

gebart, 2015-01-16:
@thomaseichinger Can it be avoided by letting the UART IRQ have a higher priority? To be able to transmit while another ISR is waiting for the lock?

haukepetersen, 2015-01-16:
@gebart: I would not start with higher IRQ priorities - as for (i) some platforms do not support this and (ii) we have not really investigated if the kernel is actually able to run with different priorities (it's still an open task...).

But I think we could add a lock for the UART, but we can just not use it for stdout...

gebart, 2015-01-16:
I got an idea: 
Using atomic counters it should be possible to implement the UART buffering functions without locking, with the drawback that we may drop bytes when the buffer gets full.

haukepetersen, 2015-01-16:
This would work if the UART driver would buffer... But it is explicitly designed not to buffer, as this is something intended to do by the higher-level modules put on top of the UART.

thomaseichinger, 2015-01-19:
I updated the description to also include parts using the updated interfaces to track adoption.

gebart, 2015-01-27:
Just for completeness I added the kinets_common stuff still being worked on in #2265

gebart, 2015-03-03:
After #2326 was merged tonight I believe this can be closed as resolved.

PeterKietzmann, 2015-03-04:
jap, correct

closed 2015-03-04 by PeterKietzmann