RIOT-OS/RIOT #2336: vtimer does not compile with clang on OS X

josephnoir, 2015-01-21:
The commit 99f227bf33834cdf7c9305865b326cfeeddfcd96 introduces a weak alias that does not work on OS X:
```
path/to/RIOT/sys/vtimer/vtimer.c:44:58: error: only weak aliases are supported on darwin
void _gettimeofday(void)           __attribute__ ((weak, alias("vtimer_gettimeofday")));
```

LudwigOrtmann, 2015-01-21:
Which compiler/board/...?

josephnoir, 2015-01-21:
Oh, sry. Apple LLVM version 6.0 (clang-600.0.56) (based on LLVM 3.5svn) for native.
Using gcc-none-eabi for the stm32f4discovery works fine.

LudwigOrtmann, 2015-01-21:
It also works on OS X when compiling for native with gcc (installed via brew).

LudwigOrtmann, 2015-01-21:
Seems to be a well known clang limitation on OSX according to the web...

LudwigOrtmann, 2015-01-21:
After looking into this a bit I can come up with no proper solution.
We could `#ifndef BOARD_NATIVE` this out for native as it's unneeded there anyways.

josephnoir, 2015-01-21:
Sounds good. Is it possible to use clang for cross compilation, and if so would this be a problem again or is it indeed OS X specific?

I searched the web a bit, but didn't find much information about it.

LudwigOrtmann, 2015-01-21:
I'm not sure about clang for arm etc, AFAIK nobody has gotten it to produce working binaries with RIOT yet. @kaspar030 ?

Either way - my proposal is a kludge for a kludge for an obsolete function... I'd rather hope we could get rid of the whole thing before we need to fiddle around with it further ;)

gebart, 2015-01-21:
@LudwigOrtmann I have built fully working contiki binaries for the Mulle using clang and I am currently working on building RIOT with clang. See also #2331

LudwigOrtmann, 2015-01-21:
@gebart Good to know! Can you shed some light on the question at hand - i.e. can the clang you use cope with `attribute__ ((weak`? Under OSX?

gebart, 2015-01-21:
@LudwigOrtmann I am using Clang 3.5 on Linux. I have used `__attribute__((weak, alias))` on arm-none-eabi, but never built anything for darwin (OSX)

The error message is rather weird, it says that _only_ weak aliases are supported, but the alias in question _is_ defined as a weak alias..

LudwigOrtmann, 2015-01-21:
The error message is off - only the `weakref` attribute is really supported. The problem now is that this only allows for static functions and therefor is useless to us in this context.

gebart, 2015-01-21:
Is it possible to solve this using some `#ifdef darwin` or similar?

LudwigOrtmann, 2015-01-21:
Well, if the problem exists in clang for arm on OSX, this won't do any good because the line actually needed on some boards ;)

gebart, 2015-01-21:
From what I can gather from posts around the web is that this error only exists when targeting darwin. i.e. It will only show up for `BOARD=native` on darwin

LudwigOrtmann, 2015-01-22:
OK then, let's kludge ;)

closed 2015-01-22 by LudwigOrtmann