RIOT-OS/RIOT #2341: buildtests: shrink the build matrix

LudwigOrtmann, 2015-01-22:
We're reaching a point where the `boards X applications` matrix lasts too long to build on Travis CI for all permutations.

Suggestions (don't get caught up on individual boards right now, I'm just trying to illustrate what I am getting at) :
- cut out  boards which don't really differ from a build point of view (e.g. only one of pca1000, pca1005, yunjia-nrf51822?)
- build drivers for only one board per CPU
- ignore old esoteric boards (e.g. pttu, wsn430-v1_4, wsn430-v1_3b)

LudwigOrtmann, 2015-01-22:
If someone can think of a way to quickly assess which boards are affected by a particular PR (so we can only build those) please speak up!

cgundogan, 2015-01-22:
>cut out boards which don't really differ from a build point of view (e.g. only one of pca1000, pca1005, yunjia-nrf51822?)

Is it possible (and does it make sense?) to have those *similar* boards picked randomly? So that we still can have a greater coverage over time

LudwigOrtmann, 2015-01-22:
Possible definitely, not sure whether it is desirable. Maybe we can have randomized equivalence-queues ;)

N8Fear, 2015-01-23:
What about actually make use of the build matrix? Take a look at http://docs.travis-ci.com/user/build-configuration/#The-Build-Matrix

I haven't thought about implementation but this would make it possible to run a Travis instance for unit tests, one for integration tests and if required even one for each board.

LudwigOrtmann, 2015-01-23:
Ah, I almost forgot about that - @Kijewski had looked into it some time ago I think.

OlegHahm, 2015-01-28:
> ignore old esoteric boards (e.g. pttu, wsn430-v1_4, wsn430-v1_3b)

I'm still against this. Either we drop the support for a board at all (might be considered for PTTU) or keep it in the testing loop. (By the way wsn430 boards are still actively used in the IoT-LAB testbed).

OlegHahm, 2015-01-28:
I would also vote for heavily cleaning the test application (and even the example) folder.

kaspar030, 2015-01-29:
Some more suggestions:
* optimize the build process more, e.g., use ccache, keep old builds and don't clean (depending on make to only rebuild changed files)
* get a seriuos build cluster and drop travis

N8Fear, 2015-01-29:
I guess "not cleaning" may lead to strange issues, as does ccache. Is it even possible to do that while using travis?

LudwigOrtmann, 2015-02-05:
ccache does not really help here I'm afraid. No (significant) amount of applications is built twice anyways.

phiros, 2015-02-06:
I played around with the matrix build feature of travis a little bit in #2394 ....
However, the results are kind of disappointing (builds often terminate prematurely etc.)

LudwigOrtmann, 2015-03-07:
https://github.com/RIOT-OS/RIOT/pull/2394 fixed this

closed 2015-03-07 by LudwigOrtmann