RIOT-OS/RIOT #2344: Linking fails with -fno-common because of multiple global variables

gebart, 2015-01-22:
Adding `-fno-common` to CFLAGS makes linking fail with the following errors:

```
.../src/riot/examples/rpl_udp/bin/mulle/rpl.a(rpl.o): In function `rpl_recv_DAO_ACK':
.../src/riot/sys/net/routing/rpl/rpl.c:441: multiple definition of `rpl_process_pid'
.../src/riot/examples/rpl_udp/bin/mulle/rpl.a(rpl_dodag.o):.../src/riot/sys/net/routing/rpl/rpl_dodag.c:39: first defined here
.../src/riot/examples/rpl_udp/bin/mulle/sixlowpan.a(ip.o): In function `byteorder_swaps':
.../src/riot/sys/net/include/sixlowpan/ip.h:425: multiple definition of `ipv6_buf'
.../src/riot/examples/rpl_udp/bin/mulle/rpl.a(rpl.o):.../src/riot/sys/net/routing/rpl/rpl.c:441: first defined here
.../src/riot/examples/rpl_udp/bin/mulle/sixlowpan.a(mac.o): In function `set_ieee802154_fcf_values':
.../src/riot/sys/net/network_layer/sixlowpan/mac.c:64: multiple definition of `msg_q'
.../src/riot/examples/rpl_udp/bin/mulle/rpl_udp.a(helper.o):.../src/riot/examples/rpl_udp/helper.c:44: first defined here
/usr/libexec/gcc/arm-none-eabi/ld: Warning: size of symbol `msg_q' changed from 256 in .../src/riot/examples/rpl_udp/bin/mulle/rpl_udp.a(helper.o) to 512 in .../src/riot/examples/rpl_udp/bin/mulle/sixlowpan.a(mac.o)
.../src/riot/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_rx.o): In function `at86rf231_rx_handler':
.../src/riot/drivers/at86rf231/at86rf231_rx.c:38: multiple definition of `buffer'
.../src/riot/examples/rpl_udp/bin/mulle/sixlowpan.a(ip.o):.../src/riot/sys/net/include/sixlowpan/ip.h:425: first defined here
/usr/libexec/gcc/arm-none-eabi/ld: Warning: size of symbol `buffer' changed from 260 in .../src/riot/examples/rpl_udp/bin/mulle/sixlowpan.a(ip.o) to 381 in .../src/riot/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_rx.o)
.../src/riot/examples/rpl_udp/bin/mulle/routing.a(etx_beaconing.o): In function `etx_get_metric':
.../src/riot/sys/net/routing/etx_beaconing.c:494: multiple definition of `tcmd'
.../src/riot/examples/rpl_udp/bin/mulle/rpl_udp.a(helper.o):.../src/riot/examples/rpl_udp/helper.c:44: first defined here
collect2: error: ld returned 1 exit status
```

The reason is that certain globals are defined in more than one compilation unit (without `extern` or `static`).

When using `-fcommon` (the default on my gcc at least), globals with the same name are merged into the same variable without signalling errors.

I believe all of these common symbols are bugs within RIOT, some should be modified to be defined `static` (e.g. `size of symbol x changed`), and some should be merged into an  `extern` declaration and a definition in a single `.c` file (e.g. `rpl_process_pid` in `sys/net/routing/rpl/rpl_dodag.c`)

```diff
diff --git a/sys/net/routing/rpl/rpl_dodag.c b/sys/net/routing/rpl/rpl_dodag.c
index cd444e8..9662eb6 100644
--- a/sys/net/routing/rpl/rpl_dodag.c
+++ b/sys/net/routing/rpl/rpl_dodag.c
@@ -31,7 +31,6 @@ char addr_str[IPV6_MAX_ADDR_STR_LEN];
 #endif
 #include "debug.h"
 
-kernel_pid_t rpl_process_pid;
 rpl_instance_t instances[RPL_MAX_INSTANCES];
 rpl_dodag_t dodags[RPL_MAX_DODAGS];
 rpl_parent_t parents[RPL_MAX_PARENTS];
```

LudwigOrtmann, 2015-01-22:
core isn't affected?

gebart, 2015-01-22:
I added the major label because I believe this can cause hard to chase down bugs with buffer corruptions as a result.

gebart, 2015-01-22:
There may be more hidden commons in the tree, I only detected these by building the rpl_udp example.

Kijewski, 2015-01-22:
Common variables are an error and against our coding conventions.

gebart, 2015-01-22:
> Common variables are an error and against our coding conventions. 

Based on this I would say that all platforms should be updated to add `-fno-common` to CFLAGS/LDFLAGS

gebart, 2015-01-22:
I did some filtering of the sources and I think I have a list of all global variables and whether they have an `extern` declaration within any header files.

https://gist.github.com/gebart/67757ad94781b2fa3391#file-extern-declarations

The commands used to generate the lists:

```sh
find -name '*.c' -exec ctags -x --c-kinds=v --file-scope=no --sort=yes {} + | awk -F ' ' '{ print $1; }' > globals.txt
for f in $(cat globals.txt|sort|uniq); do echo $f:; find -name '*.h' -exec egrep "extern .* ${f}" -l {} +; echo; done > globals-extern-declarations.txt
```

gebart, 2015-01-22:
My interpretation of @Kijewski 's comment is that any global variable defined in a .c file but without a corresponding `extern` declaration in some header file is a bug and should be changed to a `static` variable (reduce scope from global to same compilation unit only). Otherwise, an `extern` declaration must be added to some related header file.

Kijewski, 2015-01-23:
That's my interpretation of my comment, too. ;)

gebart, 2015-02-02:
Resolved the immediate bug by #2349 , #2348 and #2352 . Eliminating the rest of the unintentional common globals are continuing in #2346

closed 2015-02-02 by gebart