RIOT-OS/RIOT #2346: Tracker: Reduce scope on unintended COMMON variables

gebart, 2015-01-23:
See #2344 for a description of the error.

This is a tracker list of global variables without clear scoping (either `static` or `extern`)

The following files have global variables defined which may incorrectly get treated as COMMON symbols when linking, if any other module now or in the future defines another global with the same name. This behaviour can lead to corruption of buffers, (because two drivers have accidentally been allocated the same buffer area in RAM only because they used the same name for the variable) and a range of other problems.

The solution to these problems is to add `-fno-common` to CFLAGS for all platforms to catch these kinds of unintended aliasing and reduce the scope to `static` of any global variables that are not meant to be accessed outside of the current source file.

Below is a list of global variables that are not declared `static` and do not have a corresponding `extern` declaration in any header file. The filenames and line numbers of the definitions are shown below each variable. These variables should most likely be changed to `static`.

It is a very long list, and some of these may not be that important once `-fno-common` is in place. The most important ones are the variables which are defined in more than one file.

Line numbers are based on current master (2014.12-160-gd7eaa81)

 - [ ] abort_count
    - [ ] ./boards/msb-430h/driver_cc110x.c:27 `volatile int abort_count;`
    - [ ] ./boards/wsn430-v1_3b/driver_cc110x.c:29 `volatile int abort_count;`
    - [ ] ./boards/telosb/driver_cc2420.c:43 `volatile int abort_count;`
    - [ ] ./boards/wsn430-v1_4/driver_cc2420.c:46 `volatile int abort_count;`
    - [ ] ./boards/z1/driver_cc2420.c:57 `volatile int abort_count;`
 - [ ] abr_addr
    - [ ] ./sys/net/network_layer/sixlowpan/border/border.c:52 `ipv6_addr_t *abr_addr;`
 - [ ] abr_cache
    - [ ] ./sys/net/network_layer/sixlowpan/icmp.c:108 `ndp_a6br_cache_t abr_cache[ABR_CACHE_SIZE];`
 - [ ] abr_count
    - [ ] ./sys/net/network_layer/sixlowpan/icmp.c:100 `uint8_t abr_count = 0;`
 - [ ] abro_version
    - [ ] ./dist/tools/linux-border_router/sixlowdriver.c:46 `uint16_t abro_version = 0;`
 - [ ] adc_config
    - [ ] ./cpu/stm32f4/periph/adc.c:32 `adc_config_t adc_config[ADC_NUMOF];`
 - [ ] addr_str_mode
    - [ ] ./sys/net/routing/rpl/rpl_nonstoring/rpl_nonstoring.c:30 `char addr_str_mode[IPV6_MAX_ADDR_STR_LEN];`
    - [ ] ./sys/net/routing/rpl/rpl_storing/rpl_storing.c:31 `char addr_str_mode[IPV6_MAX_ADDR_STR_LEN];`
 - [ ] aes_interface
    - [ ] ./sys/crypto/aes.c:43 `block_cipher_interface_t aes_interface = {`
 - [ ] AIMVAL
    - [ ] ./cpu/mc1322x/maca/maca.c:806 `const uint32_t AIMVAL[19] = {`
 - [ ] appserver_stack
    - [ ] ./examples/ccn-lite-client/main.c:50 `char appserver_stack[KERNEL_CONF_STACKSIZE_MAIN];`
 - [ ] args
    - [ ] ./boards/redbee-econotag/tools/mc1322x-load.c:54 `char *args = NULL;`
 - [ ] arm_abortflag
    - [ ] ./cpu/arm7_common/bootloader.c:64 `volatile int arm_abortflag = 0;`
 - [ ] at86rf231_data_packet_cb
    - [ ] ./drivers/at86rf231/at86rf231.c:49 `netdev_rcv_data_cb_t at86rf231_data_packet_cb;`
 - [ ] at86rf231_raw_packet_cb
    - [ ] ./drivers/at86rf231/at86rf231.c:48 `netdev_802154_raw_packet_cb_t at86rf231_raw_packet_cb;`
 - [ ] baud
    - [ ] ./boards/redbee-econotag/tools/mc1322x-load.c:46 `int baud = B115200;`
 - [ ] baud_list
    - [ ] ./boards/msba2-common/tools/src/settings.c:44 `char *baud_list[] = {"115200", "57600", "38400",`
 - [ ] baud_rate
    - [ ] ./boards/msba2-common/tools/src/serial.c:56 `char *baud_rate = "115200";`
    - [ ] ./dist/tools/linux-border_router/serial.c:56 `char *baud_rate = "115200";`
 - [ ] big_buf
    - [ ] ./examples/ccn-lite-client/main.c:59 `unsigned char big_buf[3 * 1024];`
 - [ ] block
    - [ ] ./pkg/libcoap/libcoap/examples/client.c:53 `coap_block_t block = { .num = 0, .m = 0, .szx = 6 };`
 - [ ] busy
    - [ ] ./tests/irq/main.c:27 `volatile int busy, i, k;`
 - [ ] busy_stack
    - [ ] ./tests/irq/main.c:26 `char busy_stack[KERNEL_CONF_STACKSIZE_MAIN];`
 - [ ] c1
    - [ ] ./cpu/cc430/cc430-gpioint.c:47 `uint16_t c1 = 0, c2 = 0;`
 - [ ] c2
    - [ ] ./cpu/cc430/cc430-gpioint.c:47 `uint16_t c1 = 0, c2 = 0;`
 - [ ] cc1100_conf
    - [ ] ./drivers/cc110x_legacy_csma/cc1100-defaultSettings.c:61 `char cc1100_conf[] = {`
 - [ ] cc1100_mutex_pid
    - [ ] ./drivers/cc110x_legacy_csma/cc1100_phy.c:79 `volatile kernel_pid_t cc1100_mutex_pid;`
[... 790 lines cut ...]
    - [ ] ./examples/rpl_udp/udp.c:38 `char udp_server_stack_buffer[KERNEL_CONF_STACKSIZE_MAIN];`
 - [ ] udp_stack_buffer
    - [ ] ./sys/net/transport_layer/udp/udp.c:38 `char udp_stack_buffer[UDP_STACK_SIZE];`
 - [ ] VCODivF
    - [ ] ./cpu/mc1322x/maca/maca.c:858 `const uint32_t VCODivF[16] = {`
 - [ ] VCODivI
    - [ ] ./cpu/mc1322x/maca/maca.c:851 `const uint8_t VCODivI[16] = {`
 - [ ] verbose
    - [ ] ./boards/redbee-econotag/tools/mc1322x-load.c:47 `int verbose = 0;`
 - [ ] wait_seconds
    - [ ] ./pkg/libcoap/libcoap/examples/client.c:55 `unsigned int wait_seconds = 90; /* default timeout in seconds */`
 - [ ] wr_index
    - [ ] ./cpu/lpc2387/i2c/i2c.c:47 `volatile uint32_t wr_index = 0;`
 - [ ] writer
    - [ ] ./sys/net/routing/aodvv2/writer.c:40 `struct rfc5444_writer writer;`
 - [ ] x86_current_interrupt
    - [ ] ./cpu/x86/x86_interrupts.c:102 `uint8_t x86_current_interrupt;`
 - [ ] x86_current_interrupt_error_code
    - [ ] ./cpu/x86/x86_interrupts.c:103 `unsigned long x86_current_interrupt_error_code;`
 - [ ] x86_interrupted_ctx
    - [ ] ./cpu/x86/x86_interrupts.c:104 `struct x86_pushad x86_interrupted_ctx;`
 - [ ] x86_interrupt_handler_stack
    - [ ] ./cpu/x86/x86_interrupts.c:105 `char x86_interrupt_handler_stack[2048];`
 - [ ] zerolen
    - [ ] ./boards/redbee-econotag/tools/mc1322x-load.c:53 `int zerolen = 0;`

LudwigOrtmann, 2015-01-23:
Wow, wouldn't it have been just as much work to just rename them? ;)

gebart, 2015-01-23:
@LudwigOrtmann 
```sh
for f in $(cat globals.txt|sort|uniq); do find -name '*.h' -exec egrep "extern .* ${f}" -q {} + || (echo " [ ] $f" && (find -name '*.c' -exec ctags -x --c-kinds=v --file-scope=no --sort=yes {} + | grep "^${f} "| awk -F ' ' '{ printf "    [ ] %s:%d `", $4,$3; for(i=5;i<=NF;i++)printf "%s",$i (i==NF?"":OFS); printf "`\n"; }';) ) ; done > globals.md
```

LudwigOrtmann, 2015-01-23:
I just checked - `_native_argv` is declared extern in `cpu/native/include/native_internal.h`.

gebart, 2015-01-23:
There was a bug in my script. I am regenerating the list now.

gebart, 2015-01-23:
Updated. List shrunk from 864 to 720 items.

gebart, 2015-02-01:
#2352 reduces the severity of this issue by introducing error detection for unintended common variables.

cladmi, 2017-09-08:
I can try looking into this to dig a bit more in the repository. First remove easy ones, like tests, so it narrows down to the most important ones.

@gebart Can you provide the command you run to get the output please ?

What is the idea for `tunslip` tools, should they be fixed, or should they stay as close as possible to contiki version ?

What should be the solution for the `xxx_saul_driver` structures ?
I do not think they should go to the default driver implementation header, so should they have each there own header, or a common big one ?.
They are only used by `sys/auto_init/saul/auto_init_DRIVER/'.

The current situation is actually not good, for 'adc', the `const` keyword has been dropped.

This question will go for all goes

cladmi, 2017-09-08:
There may be symbols not found, like `bf` in `tests/bloom_bytes/main.c` which is defined with `BITFIELD(bf, BLOOM_BITS);`

gebart, 2017-09-09:
I will try to find what line was used to generate the globals.txt file above, the second stage of the list generation is the oneliner pasted in my comment https://github.com/RIOT-OS/RIOT/issues/2346#issuecomment-71177366

Don't bother with tunslip, it's a big mess and it is only built by itself standalone so it doesn't matter what is exported or not. 
The saul question about where the struct goes and why it's not made const is a separate issue and it would be more visible if you discuss it in a separate thread.

maribu, 2022-09-16:
Since compilation is done now with `-fno-common` (and apparently works), the issue seems to be resolved:

https://github.com/RIOT-OS/RIOT/blob/86f9d7953d2b06243ce4e3fa837c3dd21dafc9ed/makefiles/cflags.inc.mk#L60-L61

closed 2022-09-16 by maribu