RIOT-OS/RIOT #2347: Unable to use CURL in hello-world program

shanthanpalle, 2015-01-23:
Hi,
    Am trying to use curl in hello-world program,its successfully building and when i execute .elf its giving 
"Segmentation fault (core dumped)" Error at curl_global_init(CURL_GLOBAL_ALL); itself

Please get back to me soon..
OS: ubuntu 14.04
gcc -4.9

LudwigOrtmann, 2015-01-23:
curl?!

LudwigOrtmann, 2015-01-23:
Without seeing your code, I can't say anything about this.

shanthanpalle, 2015-01-23:
Here is my code

#include <stdio.h>
#include <curl/curl.h>

int main(void)
{
    CURL *curl_handle;
    CURLcode res;
    puts("Hello World!");

    printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    printf("This board features a(n) %s MCU.\n", RIOT_MCU);
    curl_global_init(CURL_GLOBAL_ALL);

    /* init the curl session */
    curl_handle = curl_easy_init();
    curl_easy_setopt(curl_handle, CURLOPT_HTTPAUTH, CURLAUTH_BASIC | CURLAUTH_DIGEST);
    curl_easy_setopt(curl_handle, CURLOPT_USERNAME, "username");
    curl_easy_setopt(curl_handle, CURLOPT_PASSWORD, "password");
    curl_easy_setopt(curl_handle, CURLOPT_UNRESTRICTED_AUTH, 1L);
    curl_easy_setopt(curl_handle, CURLOPT_NOPROGRESS, 1L);
    curl_easy_setopt(curl_handle, CURLOPT_CONNECTTIMEOUT, 30L);
    curl_easy_setopt(curl_handle, CURLOPT_TIMEOUT, 30L);                                                                                       
    return 0;
}

LudwigOrtmann, 2015-01-23:
Makefile missing.

I assume you're trying to link RIOT/native with some library on your host system?

LudwigOrtmann, 2015-01-23:
The code snippet is incomplete. Can you link to some GitHub repository instead?

shanthanpalle, 2015-01-23:
here is the source code link

https://github.com/shanthanpalle/myRIOTEx/archive/master.zip

shanthanpalle, 2015-01-23:
i have changed /RIOT-master/examples/hello-world/Makefile

shanthanpalle, 2015-01-23:
https://github.com/shanthanpalle/myRIOTEx.git

LudwigOrtmann, 2015-01-23:
Linking external libraries is not supported. What are trying to accomplish? You couldn't use curl (without porting it first) on a physical board anyways.

LudwigOrtmann, 2015-01-23:
By the way - in order to work with a repo on github you can just fork it (thereby retaining the projects commit history) with the little button on the top-right corner of a project's page.

closed 2015-01-23 by LudwigOrtmann

shanthanpalle, 2015-01-23:
Am just trying connect with my http server (like post,get etc) for that am using CURL.

LudwigOrtmann, 2015-01-23:
I suggest using a facade (mock interface) in the meantime until you or someone else adds an HTTP library for RIOT.
Please also note that the network stack is currently completely reworked. As a consequence there is currently no border router. For more information on that subject please search the mailing list archives and look at the open issues/pull requests on this site.