RIOT-OS/RIOT #2360: unknown type name: 'clockid_t' on OS X

josephnoir, 2015-01-27:
Using pthread on OS X results in this error message:

```
In file included from /Users/noir/Desktop/RIOT/tests/pthread/main.c:24:
In file included from /Users/noir/Desktop/RIOT/sys/posix/pthread/include/pthread.h:41:
/Users/noir/Desktop/RIOT/sys/posix/pthread/include/pthread_cond.h:87:63: error: unknown type name 'clockid_t'; did you mean 'clock_t'?
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/noir/Desktop/RIOT/cpu/native/periph
int pthread_condattr_getclock(const pthread_condattr_t *attr, clockid_t *clock_id);
                                                              ^~~~~~~~~
                                                              clock_t
/usr/include/sys/_types/_clock_t.h:30:33: note: 'clock_t' declared here
typedef __darwin_clock_t        clock_t;
                                ^
In file included from /Users/noir/Desktop/RIOT/tests/pthread/main.c:24:
In file included from /Users/noir/Desktop/RIOT/sys/posix/pthread/include/pthread.h:41:
/Users/noir/Desktop/RIOT/sys/posix/pthread/include/pthread_cond.h:96:57: error: unknown type name 'clockid_t'; did you mean 'clock_t'?
int pthread_condattr_setclock(pthread_condattr_t *attr, clockid_t clock_id);
                                                        ^~~~~~~~~
                                                        clock_t
/usr/include/sys/_types/_clock_t.h:30:33: note: 'clock_t' declared here
typedef __darwin_clock_t        clock_t;
                                ^
2 errors generated.
make[1]: *** [/Users/noir/Desktop/RIOT/tests/pthread/bin/native/pthread/main.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

josephnoir, 2015-03-12:
As the compiler output suggests, simply renaming all instances of `clockid_t` to `clock_t` fixes the problem on OS X. Not sure how this affects other platforms and compilers.

jhollister, 2015-03-21:
From what I can find `clock_t` is defined as a long and `clockid_t` is an int. I think all we need to fix this compiler error is to conditionally create a typedef to an int for `clockid_t` in `pthread_cond.h`.

josephnoir, 2015-03-24:
That should work too. So the fix would be to add something like the following to the file:
```C
#if defined(__APPLE__)
typedef int clockid_t;
#endif
```

BytesGalore, 2015-04-14:
@jhollister, could you test #2811? Unfortunately I have no OSX machine to test it.

josephnoir, 2015-04-21:
Should be fixed with #2811.

closed 2015-04-21 by josephnoir