RIOT-OS/RIOT #2361: malloc.h not found on OS X

josephnoir, 2015-01-27:
Using the native port on OS X with clang I get this error when using pthread:

```
/Users/noir/Desktop/RIOT/sys/posix/pthread/pthread.c:22:10: fatal error: 'malloc.h' file not found
#include <malloc.h>
         ^
1 error generated.
```

gebart, 2015-01-28:
There is one in the tree at `./sys/oneway-malloc/include/malloc.h`

The "real" posix malloc is defined in `#include <stdlib.h>`

PeterKietzmann, 2015-03-04:
@josephnoir is this fixed now? Can we close this issue?

josephnoir, 2015-03-06:
No, this is not fixed.

PeterKietzmann, 2015-03-06:
Did you test @gebart's proposal? Are you willing to fix it :-)?

jhollister, 2015-03-07:
Is there a reason to use malloc.h over stdlib.h. It's my understanding that stdlib.h is more portable so it should be fine to change all `#include <malloc.h>` to `#include <stdlib.h>` or will this cause issues somewhere?

LudwigOrtmann, 2015-03-07:
Unconditionally including `stdlib.h` leads to conflicts:
```
/home/lo/RIOT/sys/posix/pthread/include/pthread_threading_attr.h:34:3: error: conflicting types for ‘pthread_attr_t’
```

gebart, 2015-03-07:
@LudwigOrtmann is that an error in either implementation or are we just not compatible with Linux?

jhollister, 2015-03-07:
I was able to use an if define to get around the `malloc.h` issue in OSX but I still couldn't compile because of #2360

LudwigOrtmann, 2015-03-08:
@jhollister care to open a PR with your fix?

jhollister, 2015-03-12:
I don't have much experience with compatibility stuff like this. Is it correct to do something like:
```
#if defined(__MACH__)
#include <stdlib.h>
#else 
#include <malloc.h>
```
and should I do this for every occurrence of `#include <malloc.h>`?

LudwigOrtmann, 2015-03-12:
You could also try to mess around with conditionally adding a local `malloc.h` for OSX (probably by extending the include path).

LudwigOrtmann, 2015-03-21:
#2590 has been merged

closed 2015-03-21 by LudwigOrtmann