RIOT-OS/RIOT #2375: boards: unified OpenOCD script

haukepetersen, 2015-01-30:
We have a large number of boards, that are using OpenOCD as flash and debug tool. Right now these boards each contain at least 3 script files for flashing, debugging and resetting (`dist/flash.sh`, `dist/debug.sh`, `dist/reset.sh`), plus some specific GDB and OpenOCD configuration...

@authmillenon proposed to introduce a unified script, maybe something similar to the init.d scripts in Linux?
```
RIOTDIR/dist/tools/openocd/openocd.sh flash param param
```
or
```
RIOTDIR/dist/tools/openocd/openocd.sh debug param
```
and so on?

OlegHahm, 2015-01-30:
Sounds like an excellent idea to me.

authmillenon, 2015-01-30:
One problem that did arise however, how we deal with different programs (e. g. most debug.sh use openocd as a debug server while using gdb as a debug client). I. e. some parameter parsing needs to be done…

gebart, 2015-01-30:
@authmillenon How will moving the scripts from boards/x/dist to dist/tools/openocd make the argument parsing more complex?
As I understand the intention was to keep  the current DEBUGFLAGS, FFLAGS etc parts in the Makefiles of each board and move only the scripts currently (mostly) duplicated across the boards' dist directories.

authmillenon, 2015-01-30:
Sure, we can do this, and this would simplify a lot, but this is currently not the case, so I did not thought of this ;-).
While the flash and reset scripts for openocd look mostly similar for each openocd-enabled board (the given parameters tend to differ sometimes), the debug scripts seem to come in four flavors:

* openocd gets started as a server beforehand, than gdb with an exec-script.
* openocd gets started as a server beforehand, than gdb without an exec-script.
* gdb gets started with an exec-script.
* gdb gets started without an exec-script.

I don't know if the latter two are just erroneously leave out (or because they are started by the `debug-server` target) the openocd start, or if for some this is on purpose. If it is on purpose this would be another parameter. exec-scripts can be done with a check if an exec script is in a board specific location and if not this parameter can be left out.

haukepetersen, 2015-02-08:
ok, I found some time to think of this, here is a solution: #2408

haukepetersen, 2015-02-14:
yes. With #2408 merged I think we have the solution.

closed 2015-02-14 by haukepetersen