RIOT-OS/RIOT #2376: [RFC] Implement syslog protocol

authmillenon, 2015-01-30:
Currently we do all our logging/output via stdout/UART. Going more into the direction, that most devices running RIOT will interact with their environments via their sensors and actors and not through a graphical or command-line interface it might be a good idea to have some logging infrastructure, that supports the [syslog protocol](https://tools.ietf.org/html/rfc5424) (I know we did this in a very basic way for the [heise article](http://www.heise.de/developer/meldung/Sonderheft-iX-Developer-Embedded-Software-am-Kiosk-erhaeltlich-2112409.html)).

OlegHahm, 2015-01-30:
Also relates to https://github.com/RIOT-OS/RIOT/issues/84.

miri64, 2016-10-17:
I would say this is still of interest ;-)

miri64, 2019-08-10:
Closed by #11362.

miri64, 2022-04-19:
Addressed by https://github.com/RIOT-OS/RIOT/pull/15739