RIOT-OS/RIOT #2379: building error on some boards with fedora 19 gnu arm toolchain

mehlis, 2015-01-30:
```
$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-none-eabi/4.8.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../gcc-4.8-2013.11/configure --prefix=/usr --mandir=/usr/share/man --libdir=/usr/lib64 --with-pkgversion='Fedora 2013.11.24-2.fc20' --with-bugurl=https://bugzilla.redhat.com/ --enable-lto --infodir=/usr/share/info --target=arm-none-eabi --enable-interwork --enable-multilib --with-newlib --enable-languages=c,c++ --disable-nls --disable-shared --disable-threads --with-gnu-as --with-gnu-ld --with-gmp --with-mpfr --with-mpc --with-newlib --with-headers=/usr/arm-none-eabi/include --with-system-zlib
Thread model: single
gcc version 4.8.1 (Fedora 2013.11.24-2.fc20)
```

```
$ BOARD=stm32f4discovery make clean all flash term
Building application "hello-world" for "stm32f4discovery" with MCU "stm32f4".

"make" -C /home/c/git/riot/RIOT/boards/stm32f4discovery
"make" -C /home/c/git/riot/RIOT/core
"make" -C /home/c/git/riot/RIOT/cpu/stm32f4
"make" -C /home/c/git/riot/RIOT/cpu/cortex-m4_common
"make" -C /home/c/git/riot/RIOT/cpu/stm32f4/periph
"make" -C /home/c/git/riot/RIOT/drivers
"make" -C /home/c/git/riot/RIOT/sys
"make" -C /home/c/git/riot/RIOT/sys/auto_init
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-errno.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-errno.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-impure.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-impure.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-init.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-init.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-memcpy-stub.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-memcpy-stub.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-printf.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-printf.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-puts.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-puts.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-strlen.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-strlen.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-vfprintf.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-vfprintf.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-wsetup.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-wsetup.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-dtoa.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-dtoa.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-fflush.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-fflush.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-findfp.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-findfp.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-freer.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-freer.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-fvwrite.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-fvwrite.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/lib/libc.a(lib_a-fwalk.o) does not
[... cut ...]
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/libgcc.a(_divdi3.o)
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: error: /home/c/git/riot/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world.elf uses VFP register arguments, /usr/lib64/gcc/arm-none-eabi/4.8.1/libgcc.a(_udivdi3.o) does not
/usr/lib64/gcc/arm-none-eabi/4.8.1/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib64/gcc/arm-none-eabi/4.8.1/libgcc.a(_udivdi3.o)
collect2: error: ld returned 1 exit status
make: *** [all] Fehler 1
```

gebart, 2015-01-30:
At first glance it looks like you have some incompatibility between the C library and the code. I'm guessing wrong float config (-msoft-float vs -mfloat-abi=hard?) ? 
I can take a look at this on Sunday if nobody else has already fixed it.

gebart, 2015-01-31:
@mehlis What are your GCC multilib configurations?

(use `arm-none-eabi-gcc -print-multi-lib`)

```
% arm-none-eabi-gcc -print-multi-lib
.;
thumb;@mthumb
thumb/thumb2;@mthumb@march=armv7
thumb/cortex-m4;@mthumb@mcpu=cortex-m4
thumb/cortex-m4/float-abi-hard/fpuv4-sp-d16;@mthumb@mcpu=cortex-m4@mfloat-abi=hard@mfpu=fpv4-sp-d16
```

gebart, 2015-02-04:
Somewhat related: #2391

OlegHahm, 2015-02-05:
That's not the recommended toolchain for this platform. See https://github.com/RIOT-OS/RIOT/wiki/Board%3A-STM32F4discovery#supported-toolchains

closed 2015-02-05 by OlegHahm