RIOT-OS/RIOT #2391: Building inside a Docker container

gebart, 2015-02-04:
I want to make use of the Dockerfile provided by @kaspar030 in `/dist/Dockerfile` to let the entire compilation process run inside a Docker container.

Has anyone tried this?

Benefits:
 - Eliminate (most) differences between different Linux distributions
 - Easier to get started for beginners (only one command to install Docker, and one command to set up the container)
 - Specify compiler and library (newlib) versions
 - No need to fiddle with cross compiler installations

Drawbacks:
 - Takes up some extra disk space (the `debian:testing` Docker base image is 130 MB ish)
 - Hardware interaction is limited (no debuggers etc.)

My idea is to add an environment variable `BUILD_IN_DOCKER=(0|1)` and if set, let make spawn a new make inside a specified Docker container (`BUILD_DOCKER_ID?=riotbuild` or something) mapping the RIOT directory inside the Docker container so that all build artifacts are created inside the regular RIOT directory.

The only tools that will be needed on the host are the tools for interacting with hardware, such as OpenOCD for flashing, GDB for debugging, serial terminal tools, but these are readily available and GDB does not need to be installed for a specific target if using a multitarget build.

I am experimenting with the idea right now and I will open a PR later when I have had the time to try it.

kaspar030, 2015-02-04:
I used to build with the Docker container a lot, mainly for not having to compile some tolchains and for build testing with a defined base toolset. Integrating into the make system sounds like a nice idea to me!

You've probably seen it, but the Dockerfile shows an example on how to make the buildtests within a freshly spawned docker container.

I used something like
```# docker run -i -t -u $UID -v $(RIOTBUILD):/data/riotbuild riotbuild make -C examples/hello-world```
to build a specific application.

gebart, 2015-02-04:
@kaspar030 
I saw your example. I figured that the Makefile would run something like `docker run ... make blablabla`

OlegHahm, 2015-02-04:
> My idea is to add an environment variable BUILD_IN_DOCKER=(0|1) and if set, let make spawn a new make inside a specified Docker container (BUILD_DOCKER_ID?=riotbuild or something) mapping the RIOT [...]

I like the idea. If `BUILD_IN_DOCKER=1` would be the default we could probably simplify the whole getting started tutorial a lot.

kaspar030, 2015-02-04:
@OlegHahm Interesting idea... Would work only on fairly recent linux distributions, though.

OlegHahm, 2015-02-04:
For docker support? It is included in Debian stable since october 2014, Ubuntu since 2012.04, Fedora since July 2013 and is also available in Arch and Gentoo.

kaspar030, 2015-02-04:
@OlegHahm Sorry, for once I was thinking of MacOS and Windows users. :) On linux it should be fine.

OlegHahm, 2015-02-04:
According to https://docs.docker.com/installation/windows/ and https://docs.docker.com/installation/mac/ that's somehow doable. @thomaseichinger, @emmanuelsearch, can you confirm for MacOS?

thomaseichinger, 2015-02-04:
Should be doable. Docker introduces a VM though which needs to be taken into account. (`boot2docker` stuff)

thomaseichinger, 2015-02-04:
I set up Docker but the example line fails as in fails to compile

```bash
% docker run -i -t -u $UID -v $(pwd):/data/riotbuild riotbuild ./dist/tools/compile_test/compile_test.py
Building all applications in: examples
        Building application: ccn-lite-client (1/7) ..............
                failed: chronos (no linking), fox, iot-lab_M3, msb-430h (no linking), msba2, msbiot, native, redbee-econotag (no linking), samr21-xpro, telosb (no linking), wsn430-v1_3b (no linking), wsn430-v1_4 (no linking), z1 (no linking)
                failed .. retrying: avsextrem
        Building application: ccn-lite-relay (2/7) ..............
                failed: avsextrem, chronos (no linking), fox, iot-lab_M3, msb-430h (no linking), msba2, msbiot, native, redbee-econotag (no linking), samr21-xpro, telosb (no linking), wsn430-v1_3b (no linking), wsn430-v1_4 (no linking), z1 (no linking)
...
```

kaspar030, 2015-02-04:
Could you try to run manually to see what's wrong?
Something like ```# docker run -i -t -u $UID -v $(pwd):/data/riotbuild riotbuild bash``` might help.

thomaseichinger, 2015-02-04:
"Computer says no."

```bash
code/RIOT/RIOT@et-mbpr [master] % docker run -i -t -u $UID -v $(pwd):/data/riotbuild riotbuild bash
I have no name!@ef8745379d32:/data/riotbuild$ cd examples/hello-world/
I have no name!@ef8745379d32:/data/riotbuild/examples/hello-world$ make all
mkdir: cannot create directory '/data/riotbuild/examples/hello-world/bin/native/': Permission denied
Building application "hello-world" for "native" with MCU "native".

mkdir: cannot create directory '/data/riotbuild/examples/hello-world/bin/native': Permission denied
make[1]: *** [/data/riotbuild/examples/hello-world/bin/native/hello-world] Error 1
make: *** [all] Error 2
I have no name!@ef8745379d32:/data/riotbuild/examples/hello-world$
```

OlegHahm, 2015-02-04:
That's only because the poor guy doesn't have a name. (SCNR)

kaspar030, 2015-02-04:
what are the permissions of those ```/bin``` directories?

thomaseichinger, 2015-02-04:
Users UID doesn't match the directories' owner.

```bash
I have no name!@1d38880afb38:/data/riotbuild/examples/hello-world$ ls -al
total 44
drwxr-xr-x 1 1000 staff   238 Jan 27 19:39 .
drwxr-xr-x 1 1000 staff   408 Jan 30 16:51 ..
-rw-r--r-- 1 1000 staff   517 Oct 30 20:07 Makefile
-rw-r--r-- 1 1000 staff  1083 Jun 25  2014 README.md
drwxr-xr-x 1 1000 staff   408 Feb  4 15:06 bin
-rw-r--r-- 1 1000 staff 29386 Dec  1 13:19 log_t.txt
-rw-r--r-- 1 1000 staff   659 Jan 27 19:39 main.c
I have no name!@1d38880afb38:/data/riotbuild/examples/hello-world$ echo $UID
501
```

thomaseichinger, 2015-02-04:
Executing with explicitly setting the UID works.
```bash
% docker run -i -t -u 1000 -v $(pwd):/data/riotbuild riotbuild ./dist/tools/compile_test/compile_test.py
```

kaspar030, 2015-02-04:
@thomaseichinger why do the directories have uid 1000?

thomaseichinger, 2015-02-04:
@kaspar030 I don't know, isn't this the default linux UID or something? Maybe the the host's UID has to be added to the setup process?

kaspar030, 2015-02-04:
@thomaseichinger well, what does "echo $UID" say *outside* docker?

thomaseichinger, 2015-02-04:
OS X user's UID is 501 too and I'm using boot2docker yes.

kaspar030, 2015-02-04:
boot2docker needs to be configured to use the same uid, or you need to adjust permissions/umask... Should we keep this seperate from this PR?

gebart, 2015-02-04:
I created an organization for RIOT on the docker hub. Only lower case letters and numbers are allowed so I named it "riot". I'll continue this work tomorrow.

gebart, 2015-02-05:
So I have successfully built RIOT projects using Docker, however, there are some complications:

 - The user does not exist inside the container, that's why we see "I have no name!" in the prompt.
 - If not running with the `-u UID` flag, all operations will be performed as root, this is in my opinion a quite serious vulnerability if we want to let normal users run Docker containers without sudo that is not always being conveyed in Docker forums. (my normal user can execute `docker run -t -i -v /:/hostroot IMAGE` and read and write all files as `root` on the host.)
 - Passing environment variables. We can pass them one by one on the command line, or several inside a script file and source it inside the container. The `export` command in bash and zsh shows all environment variables, the only problem is: How to decide which variables should be passed and which should be stripped? e.g PATH should definitely not be passed inside since that messes up the toolchain paths, while BOARD must be passed. Then there are some platform specific variables such as MULLE_SERIAL for the mulle boards (#2282)

gebart, 2015-02-06:
I found one more difficulty with binaries built inside Docker:

```
Program received signal SIGINT, Interrupt.
0x0000155e in uart_write_blocking (uart=uart@entry=UART_0, data=83 'S') at /data/riotbuild/riotcpu/kinetis_common/uart.c:284
284	/data/riotbuild/riotcpu/kinetis_common/uart.c: No such file or directory.
```

The path to the source files in the debug info is wrong, which causes gdb to not find the file when trying to list the source line.

Is there any way to filter the debug information to modify the paths?

gebart, 2015-02-06:
one more problem:
```
kernel_init(): This is RIOT! (Version: UNKNOWN (builddir: /data/riotbuild/riotbase))
```

gebart, 2015-05-22:
closed via #2392

closed 2015-05-22 by gebart