RIOT-OS/RIOT #2395: SAMR21: UPD socket error

bapclenet, 2015-02-06:
I'm running rpl_udp example with a SAMR21 board. 
I got three boards as senders and one board as receiver.
The senders send one message a second so the receiver receives three messages a second.

The source code works fine in native and it seems to work fine with only on board as sender.
When I plug one more (so two senders), the program runs ok for a while and at some point behaves strangely.
Some time I receive an

    ERROR: recsize < 0!

which comes from

        if ((socket > MAX_SOCKETS) || (socket_base_sockets[socket - 1].socket_id == 0)) {
        return false;
    }

in sys/net/transport_layer/socket_base/socket.c, function:

    bool socket_base_exists_socket(int socket)

At this time socket was equal to 2 and socket_id to 0, next round, socket = 3 then 4

I've got this but several times.

And today, here is what I got with two senders and one receiver:

    socket = 11534388
    socket_base_sockets[socket - 1].socket_id = 64

And then the program goes to isr_hard_fault

Where does this come from? Where is socket incremented?

thomaseichinger, 2015-02-06:
@bapclenet Can I access your code somewhere? I'd expect it to be related to code alignment.

bapclenet, 2015-02-06:
Here is my branch:
https://github.com/bapclenet/RIOT/tree/samr21_socket_pb
Use the flag in rpl_udp.h to switch between sender (node) and receiver (passerelle) and to change the address of the board you flash

bapclenet, 2015-02-09:
@thomaseichinger , have you had time to have a look at?

thomaseichinger, 2015-02-09:
I gave it a try but didn't get too far yet. With the configuration below I always receive 
```bash
2015-02-09 14:53:48,020 - INFO # Error sending packet!
2015-02-09 14:53:48,023 - INFO # 1|20,366|38,894|20,458|40,254
...
```

One node with 
```C
/****** CONFIGURATION APP RPL_UDP ******/
#define PASSERELLE                  0
#define NODE                        1

#define ADDRESS_IPV6_BOARD          1
#define ADDRESS_ROOM                1
#define ADDRESS_PASSERELLE          1
/****** CONFIGURATION APP RPL_UDP ******/
```
The other
```C
/****** CONFIGURATION APP RPL_UDP ******/
#define PASSERELLE                  1
#define NODE                        0

#define ADDRESS_IPV6_BOARD          1
#define ADDRESS_ROOM                1
#define ADDRESS_PASSERELLE          1
/****** CONFIGURATION APP RPL_UDP ******/
```

bapclenet, 2015-02-09:
Change the adress IPV6 and room node to 2 otherwise it won't work

thomaseichinger, 2015-02-09:
Like this?
```C
#define PASSERELLE                  0
#define NODE                        1

#define ADDRESS_IPV6_BOARD          2
#define ADDRESS_ROOM                2
#define ADDRESS_PASSERELLE          1
```

bapclenet, 2015-02-09:
yes

thomaseichinger, 2015-02-09:
Similar problem:
```bash
2015-02-09 15:13:56,774 - INFO # 2|20,366|38,894|20,458|40,254
2015-02-09 15:13:57,779 - INFO # Error sending packet!
```

bapclenet, 2015-02-09:
if you let it run, it should connect to the router, shouldn't it?

bapclenet, 2015-02-09:
I'm going to try it again but I used the same configuration and it should work

bapclenet, 2015-02-09:
Are you using my entire RIOT branch? I get a 
    2|20,366|38,894|20,458|40,254
    Successful deliverd 78 bytes over UDP to abcd::ff:fe00:1 to 6LoWPAN

bapclenet, 2015-02-09:
Ok I got your problem, let me fix it

bapclenet, 2015-02-10:
@thomaseichinger, problem solved, you can have a try

bapclenet, 2015-02-10:
By the way, it appears that after a long time (one or two hours, I don't know yet), the boards don't send any messages. Tell me it happens on your board as well.

LudwigOrtmann, 2015-02-10:
possibly related: https://github.com/RIOT-OS/RIOT/pull/2419

bapclenet, 2015-02-10:
@LudwigOrtmann, it might be related yes.
Shouldn't I receive "ERROR: context not found" from [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/sixlowpan/lowpan.c#L1287)?
    
    if (con == NULL) {
            printf("ERROR: context not found\n");
            return;
        }

bapclenet, 2015-02-10:
@LudwigOrtmann,
I now use the mentioned PR.
That works for a while and then I fall in isr_hard_fault() with backtrace:

    #0  isr_hard_fault () at /cpu/samd21/startup.c:104
    #1  <signal handler called>
    #2  0x0000059e in sched_set_status (process=0x1 <interrupt_vector+1>, status=7) at /core/sched.c:129
    #3  0x00001134 in mutex_unlock (mutex=0x200057a8 <locks>) at /core/mutex.c:95
    #4  0x00014c4e in spi_release (dev=SPI_0) at /cpu/samd21/periph/spi.c:212
    #5  0x000131d4 in at86rf231_read_fifo (data=0x2000569f <buffer+127> "TA\210Ã\001", length=84 'T') at /drivers/at86rf231/at86rf231_spi.c:72
    #6  0x00012f6e in at86rf231_rx_handler () at /drivers/at86rf231/at86rf231_rx.c:45
    #7  0x00012602 in at86rf231_rx_irq () at /drivers/at86rf231/at86rf231.c:214
    #8  0x00005f9e in isr_eic () at /cpu/samd21/periph/gpio.c:962
    #9  <signal handler called>
    #10 0x0000147a in ?? ()

LudwigOrtmann, 2015-02-10:
@bapclenet I don't have time to help you with any of this right now - I just saw this possibly related PR (which is in master by now  - consider rebasing).

bapclenet, 2015-02-10:
@LudwigOrtmann, Sure, that's fine.

Let me know if you work on it soon.

Because I only send 3 messages a second to a samr21 board and it gets block at some point which I can't localise.

thomaseichinger, 2015-02-10:
@bapclenet This issue might be related to https://github.com/RIOT-OS/RIOT/issues/2418. Could you try again with `spi_acquire` and `spi_release` calls removed from `at86rf231_spi.c`.

bapclenet, 2015-02-10:
Thanks, yes it sounds like it's related as my backtrace goes into those functions.
I will let you know

bapclenet, 2015-02-10:
It still falls into <code>isr_hard_fault()</code>, here is the new backtrace (without <code>spi_acquire</code> and <code>spi_release</code>):

    bt
    #0  isr_hard_fault () at /cpu/samd21/startup.c:104
    #1  <signal handler called>
    #2  0x00014cd8 in spi_transfer_byte (dev=SPI_0, out=0 '\000', in=0x20008000 "") at /cpu/samd21/periph/spi.c:241
    #3  0x00014d6e in spi_transfer_bytes (dev=SPI_0, out=0x0 <interrupt_vector>, in=0x20008001 "", length=31) at /cpu/samd21/periph/spi.c:267
    #4  0x00014e1c in spi_transfer_regs (dev=SPI_0, reg=32 ' ', out=0x0 <interrupt_vector>, in=0x20007fcc "TA\210)\001", length=84) at /cpu/samd21/periph/spi.c:289
    #5  0x000131c6 in at86rf231_read_fifo (data=0x20007fcc "TA\210)\001", length=84 'T') at /drivers/at86rf231/at86rf231_spi.c:67
    #6  0x00012f84 in at86rf231_rx_handler () at /drivers/at86rf231/at86rf231_rx.c:47
    #7  0x00012602 in at86rf231_rx_irq () at /drivers/at86rf231/at86rf231.c:214
    #8  0x00005f9e in isr_eic () at /cpu/samd21/periph/gpio.c:962
    #9  <signal handler called>
    #10 0x0000147a in ?? ()

thomaseichinger, 2015-02-11:
This is a quick shot but could you try again with increased interrupt stack?

```diff
--- a/cpu/samd21/samd21_linkerscript.ld
+++ b/cpu/samd21/samd21_linkerscript.ld
@@ -40,8 +40,8 @@ MEMORY
-   small. 512 byte should be a save assumption here */
-STACK_SIZE = DEFINED(STACK_SIZE) ? STACK_SIZE : 0x200; /* 512 byte */
+   small. 768 byte should be a save assumption here */
+STACK_SIZE = DEFINED(STACK_SIZE) ? STACK_SIZE : 0x300; /* 768 bytes */
```

bapclenet, 2015-02-11:
I'm going to have a try

bapclenet, 2015-02-11:
Nope, the same, after a while, it goes to <code>isr_hard_fault()</code>

What made you think about that?

bapclenet, 2015-02-12:
Hi everyone, Kevin Roussel here. I checked Baptiste & Maxence issues on their development machines this morning, and I noticed the following:

* the SPI driver for SAMD21/SAMR21 sometimes hangs while waiting for a byte to arrive -- the RXC bit in INTFLAG register of SPI block never switches to 1. I don't really know why this occurs, but we probably should insert a "timeout" mechanism in the <code>spi_transfer_byte()</code> to abort the SPI transfer when such an issue arise, to avoid system hang, for example by returning the -1 value.
* an HardFault exception occurs randomly during execution of a UDP test code. After monitoring the backtrace when in <code>isr_hard_fault()</code>, it seems that the HardFault occurs when executing the <code>SVC 1</code> instruction, which is supposed to call the scheduler. The MCU then try to jump at the FFFFFFFE address, which causes the HardFault. It thus seems that some code corrupts the pointer to <code>SVC 1</code> handler at some point... Since modifying the Flash is normally impossible at runtime, one may suppose that some NVIC register(s) is (are) messed with. Is there any code, apart from the initialization routines, that could access this sensible area?

Hope that helps,

bapclenet, 2015-02-23:
@Thomaseichinger, have you had time to try the source code ?

bapclenet, 2015-03-09:
Anyone has got the same problem?

OlegHahm, 2015-08-17:
The old stack has been removed.

closed 2015-08-17 by OlegHahm