RIOT-OS/RIOT #2418: iot-lab_M3: hard fault with RPL_UDP example on the testbed

cgundogan, 2015-02-10:
**Sometimes** (I know @thomaseichinger loves these kinds of problems) a node on the iot-lab testbed runs into a hard fault when using the RPL_UDP example. I tried to narrow it down and I think the problem lies in the spi driver.
In `at86rf231_read_fifo()` the function `spi_release()` is called. There again, the mutex is unlocked and I believe this unlocking is the root of all problems, as it crashes on that particular line.

gebart, 2015-02-10:
I have seen an intermittent hard fault on mulle as well with the spi mutex using the at86rf231 driver in the RPL_UDP example which may be related. I have not managed to find what conditions trigger the fault, but it seems to happen after I type `node n` under some circumstances in the rpl_udp shell, not always. I have not had time to try to pin it down yet, but it looked like something corrupted the stack right before the hard fault, maybe some buffer overflowed?

thomaseichinger, 2015-02-10:
I think this strange behaviour results from the fact the current driver still reads the FIFO in interrupt context. I wasn't able to track this down yet as when I step through with the debugger this simply doesn't happen. I also have the rpl_udp example running now for about an hour with 5 nodes without any crash so it is really hard to narrow it down. BTW I will refactor this driver tomorrow. (Without major SPI transfers in interrupt context. Fingers crossed.)

gebart, 2015-02-10:
@thomaseichinger I will try to find the time to do some trial runs for before-and-after your changes to the rf2xx SPI transfers when you open the PR to give you feedback on this Heisenbug.

gebart, 2015-02-13:
Is it possible to see the interrupt stack usage similar to what is shown by `ps` when creating threads with `CREATE_STACKTEST`?

gebart, 2015-02-13:
In response to my own question:
I did a test where I fill the stack with a canary early after reset before libc init to measure the maximum stack depth used. Everything works now and I'm using 384 bytes out of 512 total after having run `init n` from the RPL shell.

I realized that my previous build was using a newlib built with `-fstack-protector` which makes it use more stack space for certain functions such as `printf` and `memcpy` which probably made it overflow the ISR stack. I have not tested building with that checked newlib build since I introduced the stack measurement code.

cgundogan, 2015-02-14:
@thomaseichinger I can also test your refactored driver for this board. Is there a PR anywhere?

gebart, 2015-02-14:
I have not yet tested the refactored driver, the above stack tests were made using the original driver.

thomaseichinger, 2015-02-14:
@gebart To conclude, removing `-fstack-protector` made it work for you?
@cgundogan The refactoring isn't done yet as I wanted to wait for packet buffer implementation to be stable. (https://github.com/RIOT-OS/RIOT/pull/2285)
I will ping you once there is a PR.

gebart, 2015-02-14:
@thomaseichinger Yes, everything is working properly on k60/mulle after relinking with newlib `-fno-stack-protector`.
It is important to note here that it was newlib, not RIOT, that was built with `-fstack-protector`/`-fnostack-protector` which makes several of the standard library functions for string handling use more stack space, e.g. memcpy, printf, strcat etc..

gebart, 2015-03-09:
I am getting hard faults in spi_acquire inside at86rf231_read_fifo. The locks array has been corrupted, the first mutex has the value 1 (locked, looks fine), but the pointer to the next thread waiting has been changed to 0x0020000e, and the write seems to have overflowed into the next mutex as well, it has the value 0x0020000f. Do these values have any meaning which may help us finding where this comes from? 

From the looks of the memory contents it is not simply a memcpy that has gone haywire and overwritten large chunks of memory, the rest of the memory contents looks plausible, but I have not yet tried to find a packet inside the buffer area to see if that looks fine..

An excerpt from the memory map is below:
```
 .bss.buffer    0x00000000200080da      0x17d .../RIOT/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_rx.o)
 .bss.rx_buffer_next
                0x0000000020008257        0x1 .../RIOT/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_rx.o)
                0x0000000020008257                rx_buffer_next
 .bss           0x0000000020008258        0x0 .../RIOT/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_tx.o)
 .bss.sequence_nr
                0x0000000020008258        0x1 .../RIOT/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_tx.o)
 .bss.wait_for_ack
                0x0000000020008259        0x1 .../RIOT/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_tx.o)
 .bss           0x000000002000825a        0x0 .../RIOT/examples/rpl_udp/bin/mulle/at86rf231.a(at86rf231_spi.o)
 .bss           0x000000002000825a        0x0 .../RIOT/examples/rpl_udp/bin/mulle/core.a(reboot.o)
 .bss           0x000000002000825a        0x0 .../RIOT/examples/rpl_udp/bin/mulle/cortex-m4_common.a(reboot_arch.o)
 .bss           0x000000002000825a        0x0 .../RIOT/examples/rpl_udp/bin/mulle/ieee802154.a(ieee802154_frame.o)
 .bss.ieee802154_payload_len
                0x000000002000825a        0x2 .../RIOT/examples/rpl_udp/bin/mulle/ieee802154.a(ieee802154_frame.o)
                0x000000002000825a                ieee802154_payload_len
 .bss.ieee802154_hdr_ptr
                0x000000002000825c        0x1 .../RIOT/examples/rpl_udp/bin/mulle/ieee802154.a(ieee802154_frame.o)
                0x000000002000825c                ieee802154_hdr_ptr
 .bss.ieee802154_payload_ptr
                0x000000002000825d        0x1 .../RIOT/examples/rpl_udp/bin/mulle/ieee802154.a(ieee802154_frame.o)
                0x000000002000825d                ieee802154_payload_ptr
 .bss           0x000000002000825e        0x0 .../RIOT/examples/rpl_udp/bin/mulle/kinetis_common.a(spi.o)
 *fill*         0x000000002000825e        0x2 
 .bss.locks     0x0000000020008260       0x10 .../RIOT/examples/rpl_udp/bin/mulle/kinetis_common.a(spi.o)
 .bss           0x0000000020008270        0x0 .../RIOT/examples/rpl_udp/bin/mulle/net_help.a(net_help.o)
 .bss           0x0000000020008270        0x0 .../RIOT/examples/rpl_udp/bin/mulle/net_help.a(inet_pton.o)
 .bss           0x0000000020008270        0x0 .../RIOT/examples/rpl_udp/bin/mulle/netdev_802154.a(802154.o)
 .bss           0x0000000020008270        0x0 .../RIOT/examples/rpl_udp/bin/mulle/netdev_base.a(base.o)
 .bss           0x0000000020008270        0x0 .../RIOT/examples/rpl_udp/bin/mulle/ps.a(ps.o)
 .bss           0x0000000020008270        0x0 .../RIOT/examples/rpl_udp/bin/mulle/routing.a(etx_beaconing.o)
 .bss.msg_que   0x0000000020008270      0x400 .../RIOT/examples/rpl_udp/bin/mulle/routing.a(etx_beaconing.o)
 .bss.own_address
                0x0000000020008670        0x4 .../RIOT/examples/rpl_udp/bin/mulle/routing.a(etx_beaconing.o)
```

Memory contents at the point of the crash:
```
0x20008080 <at86rf231_rx_buffer+24>:    0x40132846      0x80ea002d      0x02502000      0x01612801
0x20008090 <at86rf231_rx_buffer+40>:    0x01000000      0x29030002      0xffff7707      0x00000000
0x200080a0 <at86rf231_rx_buffer+56>:    0x00000000      0x28464e47      0x00104013      0x20008169
0x200080b0 <at86rf231_rx_buffer+72>:    0xfb01040a      0x00000161      0x01020100      0x77075c03
0x200080c0 <at86rf231_rx_buffer+88>:    0x0000ffff      0x00000000      0x4e470000      0x40132846
0x200080d0 <at86rf231_rx_buffer+104>:   0x81e8002d      0x01502000      0x4161f801      0x07775fd8
0x200080e0 <buffer+6>:  0x2d00ffff      0x46281340      0x3b7a474e      0x019b1a3a
0x200080f0 <buffer+22>: 0xf01e9290      0x0010d601      0x01200000      0x00ffd816
0x20008100 <buffer+38>: 0x00004b86      0x00000000      0x0e040100      0x0a0c0800
0x20008110 <buffer+54>: 0x00010007      0xff000100      0x1e08ffff      0x00004040
0x20008120 <buffer+70>: 0x00000000      0x00000000      0x01200000      0x00ffd816
0x20008130 <buffer+86>: 0x00004b86      0x00000000      0x00280000      0x00000000
0x20008140 <buffer+102>:        0x00000000      0x00000000      0x00000000      0x00000000
0x20008150 <buffer+118>:        0x00000000      0x00000000      0xc8411b00      0xff077729
0x20008160 <buffer+134>:        0x401000ff      0x4e462813      0x3a3b7a47      0xb8009b1a
0x20008170 <buffer+150>:        0xfb00006a      0x007c10a4      0x16012000      0x8600ffd8
0x20008180 <buffer+166>:        0x0000004b      0x00000000      0x000e0401      0x070a0c08
0x20008190 <buffer+182>:        0x00000100      0xffff0001      0x401e08ff      0x00000040
0x200081a0 <buffer+198>:        0x00000000      0x00000000      0x16012000      0x8600ffd8
0x200081b0 <buffer+214>:        0x0000004b      0x00000000      0x00009a00      0x00000000
[... cut ...]
0x20008220 <buffer+326>:        0x00000000      0x00000000      0xd8160120      0x4b8600ff
0x20008230 <buffer+342>:        0x00000000      0x00000000      0x000000f8      0x00000000
0x20008240 <buffer+358>:        0x00000000      0x00000000      0x00000000      0x00000000
0x20008250 <buffer+374>:        0x00000000      0x01000000      0x07000007      0x00000000
0x20008260 <locks>:     0x00000001      0x0020000e      0x0020000f      0x00000000
0x20008270 <msg_que>:   0x00000000      0x00000000      0x00000000      0x00000000
```
`msg_que` is all zeroes onward.

gebart, 2015-03-09:
The hard fault only appears after a long uptime and with traffic going between other nodes in the 802.15.4 network

gebart, 2015-03-10:
Sorry for hijacking the thread, my issues may not be related to the original report. I need to investigate further..

cgundogan, 2015-04-07:
@thomaseichinger is this issue fixed?

cgundogan, 2015-08-17:
I close this issue, because its content is highly depending on the old network stack and drivers. Furthermore, I could not see similar hard faults with the `ng_networking` example on the `iot-lab_M3`.

closed 2015-08-17 by cgundogan