RIOT-OS/RIOT #2422: RFC: doc: Common doxygen naming scheme for modules.

authmillenon, 2015-02-10:
This is a RFC to find a common doxygen naming scheme for (sub-)modules and add it to the coding guidelines. Some examples where this currently is not the case:

* `board_spark-core` <=> `boards_telosb`
* `native_cpu` <=> `lpc2387` <=> `cpu_atmega2560`
* `diskio` <=> `driver_isl29020` <=> `drivers_mag3110`
* `sys_vtimer` <=> `net` <=> `net_datalink` <=> `tcp` <=> `net_sixlowpan`

This can be a pain if you want to reference other modules in your documentation.
I would propose, as this is the most collision free solution and also mostly already in place to chain the modules hierarchically. With the above examples this would be:

* `boards_spark-core`
* `boards_telosb`
* `cpu_native`
* `cpu_lpc2387`
* `cpu_atmega2560`
* `drivers_diskio`
* `drivers_isl29020`
* `drivers_mag3110`
* `sys_vtimer`
* `sys_net`
* `sys_radio`
* `sys_net_tcp`
* `sys_net_sixlowpan`

LudwigOrtmann, 2015-02-10:
Sounds good.

PeterKietzmann, 2015-02-10:
I'm not against unifying the naming schemes.

authmillenon, 2015-02-10:
@PeterKietzmann But your not in favor either?

PeterKietzmann, 2015-02-10:
In general I like your (new) proposal but after looking at @haukepetersen comment in #2404 I also feel attached to his suggestion :/ Can't decide at this moment

authmillenon, 2015-02-10:
@haukepetersen's proposal would still be more or less incorparated into this one: `net_netapi` (@haukepetersen's proposal) vs `sys_net_netapi` (this issue) vs `sys_netapi` (which I proposed in #2404 before opening this issue, which triggered @haukepetersen's response) ;-)

haukepetersen, 2015-02-10:
@PeterKietzmann: I make it easy for you: +1 for @authmillenon's proposoal!

PeterKietzmann, 2015-02-11:
So the majority is pro :-)

gebart, 2015-02-11:
:+1: for @authmillenon 's suggestion

haukepetersen, 2015-02-11:
alright then, so `sys_net_somethingsomething` it is for the network stack stuff -> so lets make sure the stuff we newly introduce from now on fits into this naming scheme.

authmillenon, 2015-02-11:
:+1: who fixes the old stuff?

haukepetersen, 2015-02-12:
anyone bored to death? :-)

haukepetersen, 2015-02-12:
actually I am just doing the fixing for the boards folder (as it was me who broke many of them...)

authmillenon, 2015-02-12:
Remember to look after `@ref` tags that reference the modules you change ;-)

haukepetersen, 2015-02-12:
just stumbled upon something: after this convention the network modules should be named `sys_net_SOMETHING`. But `net` is it's own top-level doxygen group (although the files reside in sys). Do we make an exception for network modules and put them in the `net_SOMETHING` scheme? I think in this case that makes sense?!

LudwigOrtmann, 2015-02-12:
makes sense

authmillenon, 2015-02-12:
Just for consistency (and because I already did it for most of my newer PRs) I would stick to `sys_net` and `sys_net_<module>`. Why should it make ab exception?

LudwigOrtmann, 2015-02-12:
Because 'net' is kind of special in an Internet OS ;)

authmillenon, 2015-02-12:
I thought we are a real-time OS :P

LudwigOrtmann, 2015-02-13:
That's not exclusive ;)

OlegHahm, 2015-02-13:
I would also vote for `net_` instead of `sys_net_` - and I wouldn't call it an exception. Folder and doxygen hierarchy don't have to be identical.

miri64, 2016-10-17:
I think This still needs fixing.

PeterKietzmann, 2017-11-27:
It seems there are still occurrences of this inconsistency. I think a checklist would speed up progress of addressing this issue

smlng, 2018-10-08:
seems to be resolved?

miri64, 2018-10-08:
Aye

closed 2018-10-08 by miri64