RIOT-OS/RIOT #2432: Problems with packet processing in coexistence with other ieee802154 devices

jfischer-phytec-iot, 2015-02-11:
I have a ieee802154 device regularly sends a beacon request.
After a few packets the transceiver buffer runs full and is not processed further. See dump:

2015-02-11 11:21:52,952 - INFO # Packet received
2015-02-11 11:21:52,955 - INFO # transceiver: Handling KW2XRF packet
2015-02-11 11:21:52,957 - INFO # transceiver: Content: 
2015-02-11 11:21:52,960 - INFO # transceiver: Notify thread 5
2015-02-11 11:21:52,963 - INFO # transceiver: Waiting for next message
2015-02-11 11:21:52,968 - INFO # INFO: Received IEEE 802.15.4. packet (length = 1):
....
2015-02-11 11:22:23,229 - INFO # INFO: Received IEEE 802.15.4. packet (length = 1):
2015-02-11 11:22:23,230 - INFO # INFO: FCF:
2015-02-11 11:22:23,230 - INFO # frame type: 03
2015-02-11 11:22:23,233 - INFO # security enabled: 00
2015-02-11 11:22:23,234 - INFO # frame pending: 00
2015-02-11 11:22:23,236 - INFO # ack requested: 00
2015-02-11 11:22:23,238 - INFO # pan id compression: 00
2015-02-11 11:22:23,240 - INFO # destination address mode: 02
2015-02-11 11:22:23,242 - INFO # frame version: 00
2015-02-11 11:22:23,244 - INFO # source address mode: 00
2015-02-11 11:22:23,247 - INFO # Sender:00 00 00 00 00 00 00 00 
2015-02-11 11:22:23,250 - INFO # Receiver:ff ff 00 00 00 00 00 00 
2015-02-11 11:22:23,251 - INFO # Payload:
2015-02-11 11:22:23,251 - INFO # 07 
2015-02-11 11:22:23,255 - INFO # Unknown IEEE 802.15.4 source address mode.
2015-02-11 11:22:38,342 - INFO # transceiver: Transceiver: Message received, type: 06
2015-02-11 11:22:38,344 - INFO # Packet received
2015-02-11 11:22:38,348 - INFO # transceiver: buffer size exceeded, dropping packet
2015-02-11 11:22:38,351 - INFO # transceiver: Notify thread 5
2015-02-11 11:22:38,354 - INFO # transceiver: failed to notify upper layer.
2015-02-11 11:22:38,358 - INFO # transceiver: Waiting for next message

I have examined it briefly. Is there a missing ``` p->processing--; ``` ?

https://github.com/RIOT-OS/RIOT/blob/master/sys/net/network_layer/sixlowpan/mac.c#L138

jfischer-phytec-iot, 2015-02-11:
#2434 fixes it.

thomaseichinger, 2015-02-11:
#2434 seems correct. But there seems to be a problem in your FCF. :wink:

closed 2015-02-11 by jfischer-phytec-iot