RIOT-OS/RIOT #2438: ubjson: Invalid reads/writes

authmillenon, 2015-02-11:
Valgrind is noticing some invalid reads/writes in ubjson (commit 2c703e5):

```
$ make -C tests/unittests/ all-valgrind term-valgrind
[…]
valgrind --track-origins=yes --fullpath-after=/home/martine/Repositories/RIOT-OS/RIOT/ --read-var-info=yes /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/unittests.elf 
==15267== Memcheck, a memory error detector
==15267== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==15267== Using Valgrind-3.10.0.SVN and LibVEX; rerun with -h for copyright info
==15267== Command: /home/martine/Repositories/RIOT-OS/RIOT/tests/unittests/bin/native/unittests.elf
==15267== 
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

kernel_init(): This is RIOT! (Version: 2014.12-358-g10022-pktbuf/api/use-pkt)
kernel_init(): jumping into first task...
.............................................................................................
.............................................................................................
...............................................................
==15267== Invalid write of size 4
==15267==    at 0x40A5C68: makecontext (/build/buildd/eglibc-2.19/stdlib/../sysdeps/unix/sysv/linux/i386/makecontext.S:82)
==15267==    by 0x804B7DB: thread_stack_init (cpu/native/native_cpu.c:131)
==15267==    by 0x8049906: thread_create (core/thread.c:188)
==15267==    by 0x80680F2: test_ubjson_test (tests/unittests/tests-ubjson/tests-ubjson.c:89)
==15267==    by 0x8068CC7: test_ubjson_empty_object (tests/unittests/tests-ubjson/test-ubjson-empty-object.c:129)
==15267==    by 0x806C252: TestCase_run (sys/embunit/TestCase.c:58)
==15267==    by 0x806C1CF: TestCaller_run (sys/embunit/TestCaller.c:54)
==15267==    by 0x806C55B: TestRunner_runTest (sys/embunit/TestRunner.c:99)
==15267==    by 0x8068169: tests_ubjson (tests/unittests/tests-ubjson/tests-ubjson.c:114)
==15267==    by 0x804C6EB: main (tests/unittests/main.c:30)
==15267==  Location 0x809914c is 0 bytes inside receiver_stack[7788],
==15267==  a global variable declared at tests-ubjson.c:32
==15267== 
==15267== Invalid write of size 4
==15267==    at 0x40A5BE7: setcontext (/build/buildd/eglibc-2.19/stdlib/../sysdeps/unix/sysv/linux/i386/setcontext.S:73)
==15267==  Location 0x8099148 is 0 bytes inside receiver_stack[7784],
==15267==  a global variable declared at tests-ubjson.c:32
==15267== 
==15267== Invalid read of size 4
==15267==    at 0x40A5BFD: setcontext (/build/buildd/eglibc-2.19/stdlib/../sysdeps/unix/sysv/linux/i386/setcontext.S:92)
==15267==    by 0x8068036: test_ubjson_read_fun (tests/unittests/tests-ubjson/tests-ubjson.c:64)
==15267==    by 0x40A5C6A: makecontext (/build/buildd/eglibc-2.19/stdlib/../sysdeps/unix/sysv/linux/i386/makecontext.S:87)
==15267==    by 0x8087F07: ???
==15267==  Location 0x8099148 is 0 bytes inside receiver_stack[7784],
==15267==  a global variable declared at tests-ubjson.c:32
==15267== 
==15267== Invalid read of size 4
==15267==    at 0x8068087: test_ubjson_receiver_trampoline (tests/unittests/tests-ubjson/tests-ubjson.c:77)
==15267==    by 0x40A5C6A: makecontext (/build/buildd/eglibc-2.19/stdlib/../sysdeps/unix/sysv/linux/i386/makecontext.S:87)
==15267==    by 0x8087F07: ???
==15267==  Location 0x809914c is 0 bytes inside receiver_stack[7788],
==15267==  a global variable declared at tests-ubjson.c:32
==15267== 

OK (249 tests)
XXX: lpm_set(): LPM_POWERDOWN not implemented
lpm_set(): exit()
==15267== 
==15267== HEAP SUMMARY:
==15267==     in use at exit: 0 bytes in 0 blocks
==15267==   total heap usage: 17 allocs, 17 frees, 1,667 bytes allocated
==15267== 
==15267== All heap blocks were freed -- no leaks are possible
==15267== 
==15267== For counts of detected and suppressed errors, rerun with: -v
==15267== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
make: Leaving directory `/home/martine/Repositories/RIOT-OS/RIOT/tests/unittests'
```

LudwigOrtmann, 2015-02-11:
Tracked in https://github.com/RIOT-OS/RIOT/issues/2175 already.

closed 2015-02-11 by LudwigOrtmann

authmillenon, 2015-02-11:
Ups… it was already by me. Sorry.