RIOT-OS/RIOT #2470: Flashing is not working on SAMR21 any more

OlegHahm, 2015-02-19:
As stated in https://github.com/RIOT-OS/RIOT/pull/2408#issuecomment-75021353 `make flash` (using the unified OpenOCD script from #2408) is currently not working for the SAMR21 - at least on ArchLinux. (Neither with the AUR version of OpenOCD nor with a fresh compiled from current git HEAD.) Putting an `exit` at the end of the `program` command or adding a `shutdown` command at the end helps, but leads to `1` as an exit code indicating an error - but I don't know why, because flashing seems to work correctly.

OlegHahm, 2015-02-19:
@thomaseichinger, the full output with `-d3` is very long. The last lines look like
```
User : 29351 21254 command.c:763 jim_echo(): ** Verified OK **
Debug: 29352 21254 command.c:145 script_debug(): command - ocd_command ocd_command type ocd_shutdown
Debug: 29353 21255 command.c:145 script_debug(): command - shutdown ocd_shutdown
Debug: 29354 21255 adi_v5_cmsis_dap.c:129 cmsis_dap_queue_ap_read(): reg = 16
Debug: 29355 21256 adi_v5_cmsis_dap.c:161 cmsis_dap_run():  
Debug: 29356 21256 adi_v5_cmsis_dap.c:83 cmsis_dap_queue_dp_read(): reg = 4
User : 29358 21256 server.c:588 handle_shutdown_command(): shutdown command invoked
.-(~/git/RIOT/tests/bloom_bytes)-------------------------------------------------------------------------------(oleg@zoidberg)-
`--> echo $?
1      
```

haukepetersen, 2015-02-19:
to resume the discussion from #2408: i removed the `-c shutdown` after `-c reset run`, as it lead to duplicate shutdown on most boards. As example the `stm32f4discovery` with `-c shutdown`:
```
...
** Verified OK **
shutdown command invoked
shutdown command invoked
```
without `-c `shutdown`:
```
...
** Verified OK **
shutdown command invoked
```

@thomaseichinger testsed for the samr21 - does openOcd maybe behave different between Linux and OSX?

OlegHahm, 2015-02-19:
Does it hurt to shutdown twice?

LudwigOrtmann, 2015-02-19:
Can't we pass arguments/additional commands nowadays?

haukepetersen, 2015-02-19:
> Does it hurt to shutdown twice?

It's does not really that I know of, but its not clean either...

By the way, this is the output when flashing the samr21-xpro without the `-c shutdown` on my computer:
```
...
** Programming Finished **
** Verify Started **
verified 49832 bytes in 4.127607s (11.790 KiB/s)
** Verified OK **
shutdown command invoked
```

So it seems it has something to do with different OpenOCD versions? My version is `Open On-Chip Debugger 0.9.0-dev-00186-g30203b3 (2014-11-18-16:43)` build from their sources.

haukepetersen, 2015-02-19:
just updated openOCD -> `Open On-Chip Debugger 0.9.0-dev-00301-g986102f (2015-02-19-16:37)`

now I have the same behavior as @OlegHahm. Also flashing the stm32f4discovery seems to be broken: `Error: jtag status contains invalid mode value - communication failure` after successful verification...

OlegHahm, 2015-02-19:
and via versa: reverted to OpenOCD version 0.9.0-dev-00186-g30203b3 and it works like charm again.

LudwigOrtmann, 2015-02-19:
:banana:

haukepetersen, 2015-02-19:
ok, my fault. Seems like OpenOCD changed something. I guess #2471 should fix it for everyone?

gebart, 2015-02-19:
>Can't we pass arguments/additional commands nowadays?

Only extra initialization since the OPENOCD_EXTRA_INIT argument is put before all of the commands on the command line.

haukepetersen, 2015-02-19:
and the forget about the error for the stm board stated above, that was because I forgot to add the backshlash after the `-c reset halt` command...

gebart, 2015-02-23:
@LudwigOrtmann @haukepetersen How about we add argument handling to the script?
That will allow us to properly quote the arguments which is not possible via environment variables unless we use some `eval` function or similar on the variables.

LudwigOrtmann, 2015-02-23:
+1 for argument handling

PeterKietzmann, 2015-03-04:
@gebart is this issue fixed in #2487? Can we close it?

gebart, 2015-03-04:
@PeterKietzmann I don't have a SAMR21 board to test with.
@LudwigOrtmann : Can you verify this is working now?

PeterKietzmann, 2015-03-04:
I do have one but didn't want to go into detail here. AFAIK just flashing works, will verify soon. What should I especially have a look at/test?

OlegHahm, 2015-03-04:
Before #2471 or #2472 are merged the problem isn't solved.

PeterKietzmann, 2015-03-04:
Ok, good that these PRs are referenced now so we may not forget to close the issue after merging

PR #2471 (2015-03-05): dist/openocd.sh: added shutdown to flash commands
fixes #2470
closed 2015-03-05 by OlegHahm