RIOT-OS/RIOT #2479: What's wrong with tests/bitarithm_timings?

authmillenon, 2015-02-20:
Observing the Pull Requests that Travis aborted seemingly randomly building, I reallized, that most (if not all) of them halted while building `tests/bitarithm_timings` and I would guess that it's always the same board cause. So what is wrong with this test that its build causes Travis to crash?

phiros, 2015-02-22:
I think that has more to do with travis internals than with the test itself. I've also noticed this issue
quite some time ago and investigated it a little bit. To me it looks like there is nothing special about this test other than that it gets build around the 10 minute mark by travis...

BTW: I've not seen this happening with #2394.

phiros, 2015-03-22:
@authmillenon Did you see this happening with any build in the past few days?
If so: please list a few links to PRs where this happened.

authmillenon, 2015-03-22:
I'll try to look out for it.

phiros, 2015-04-28:
@authmillenon have you seen any of those issues in the past month?
If not I'd like to close this as a Memo.

authmillenon, 2015-04-28:
Not that I know of.

closed 2015-04-28 by phiros