RIOT-OS/RIOT #2486: Travis: Add separate build matrix item for static code checks

gebart, 2015-02-23:
The "needs-squashing" check is run after the compile tests have been run. Since #2394 was merged this means that if your build needs squashing all of the matrix build fail, and the person checking the Travis results need to go into each of the sub-builds and check if the failure was because of some problem, or if it was just "needs squashing" or "new doxygen warnings" or some other static check.

I propose moving all of these static checks to its own sub and let them run in parallel to the MCU group builds. This way we will get a nice overview on the main landing page for the build and we only need to open build logs for "true" failures.

I do not have any experience in configuring Travis and I do not have any time to learn more about Travis in the near future, hence only an Issue but no PR for this feature.

An example build where you have to open each sub build to check if it is a build failure or a static code check failure: https://travis-ci.org/RIOT-OS/RIOT/builds/51796761

LudwigOrtmann, 2015-02-23:
I am in favor of having static tests in parallel to the `BUILDTEST_MCU_GROUP` tests.

gebart, 2015-02-24:
Implemented in #2488

closed 2015-02-24 by gebart