RIOT-OS/RIOT #2490: RPL control messages have the wrong destination address

gebart, 2015-02-25:
RPL messages (DIO, DAO etc.) are being sent to FF02::1 instead of FF02:1A. This results in Contiki and RIOT nodes ignoring messages from each other and never joining the same DODAG.

According to RFC6550:

http://tools.ietf.org/html/rfc6550#section-6: 
> Section 6. ICMPv6 RPL Control Message
...
> Most RPL control messages have the scope of a link.  The only
> exception is for the DAO / DAO-ACK messages in Non-Storing mode,
> which are exchanged using a unicast address over multiple hops and
> thus uses global or unique-local addresses for both the source and
> destination addresses.  For all other RPL control messages, the
> source address is a link-local address, and the destination address
> is either the all-RPL-nodes multicast address or a link-local unicast
> address of the destination.  The all-RPL-nodes multicast address is a
> new address with a value of ff02::1a.

cgundogan, 2015-02-25:
Hmm right.. could you try using the patch #2491 ?
However, I have to be really optimistic to expect both implementations Contiki vs RIOT work out of the box together. Good thing though that you tested this, brings us once step further to RFC compliancy (:

gebart, 2015-03-04:
Closed via #2491

closed 2015-03-04 by gebart