RIOT-OS/RIOT #2526: routing: complete documentation in etx_beaconing.h

authmillenon, 2015-03-03:
`sys/net/include/etx_beaconing.h` needs some care.

kausthub, 2015-03-06:
hi , i am interested in completing this issue , but i am new to this , so what do you meant by needs some extra care ? 
I am sorry if that question was trivial , thanks in advance

authmillenon, 2015-03-06:
Welcome to RIOT,
as you can see https://github.com/RIOT-OS/RIOT/blob/master/sys/net/include/etx_beaconing.h does not have any doxygen documentation currently and most of the comments are not that helpful either. Most comments must only be reformated to doxygen syntax, however.

BigDaddyD, 2015-03-09:
Is there a doxygen reference here on the wiki? I can't seem to find one

gebart, 2015-03-09:
See also #2561

authmillenon, 2015-03-10:
@BigDaddyD not in the Wiki, but it's generated regularly from the source code at http://riot-os.org/api/

OlegHahm, 2016-03-09:
If I'm not mistaken this file has gone.

closed 2016-03-09 by OlegHahm