RIOT-OS/RIOT #2528: Bus initialization procedure

gebart, 2015-03-04:
Some (most) of the drivers for i2c connected and SPI connected devices initialize the bus as a step in their own initialization procedure. I think this is bad practise, at least when more than one device is connected to the same bus.

In my opinion, it only makes sense to initialize the bus in the device driver for the bus slave if the slave device is the only device on the bus, but generally you can connect many devices to the bus and you (as application designer) need to decide on the greatest common subset of parameters which can be combined by looking over all of the bus slaves' parameters with regards to timing, polarity etc.

If this is not properly handled, some bus slaves may just not work at all (best case), or break, or it may bring down the entire bus, or seem fine on the development board, but the same application won't work on a customer's board (very difficult to debug).

I suggest that bus initialization (`spi_init_master`, `i2c_init_master`) should not be called from the bus slave initialization (e.g. `mag3110_init` etc.), and instead be called with a proper set of parameters for the application, before any bus slaves are initialized.

I don't think there is any non-convoluted way of automatically computing the best available parameters given a set of slave devices because of the great variation in different parameters, some devices have both minimum and maximum timings, some can work at either polarity, some can handle high baud rates but need a long idle time between each byte transferred etc. etc.. I think it is best if human application designers handle the bus parameters for the time being.

Opinions on this?

gebart, 2015-03-04:
@PeterKietzmann , @thomaseichinger , @jfischer-phytec-iot you have been active in device driver implementation lately, what are your thoughts on this?

PeterKietzmann, 2015-03-04:
@gebart. I agree with you. Let me just list what came through my mind when thinking about this as a basis to discuss:

- we could stay with this solution and add peripheral configuration fields to the slave's init function to guaranty that all bus devices get the same parameters from an application (may blow up the slave's interface?)
- we could change to your proposed solution. Then we should introduce some kind of "`spi initialized`" flag or "`spi_is_active`" function to the peripheral driver so the slave device can check whether the peripheral bus is already initialized and otherwise throw an error
- don't know if I remember correct but did @haukepetersen had some arguments against a structure like that?

PeterKietzmann, 2015-03-04:
In addition: When we decide to adapt the initialization procedure I assume this would be "a lot" of  routine work. Could this be a Newbie Task?

gebart, 2015-03-04:
@PeterKietzmann The work will be quite routine, mostly deleting the init_master line from the driver and adding it back in tests/driver_xxx/main.c

When we reach a consensus on this and if the decision is to change the existing implementations I will create a tracker issue for all drivers which need modifying.

PeterKietzmann, 2015-03-04:
Jupp. I'd like to hear at least one more opinion on that. This is a good newbie task

thomaseichinger, 2015-03-04:
For the new network stack we will have to introduce an initialisation procedure which enables an application designer to granularly choose what to use. This could then be combined with initialising buses or other shared resources.

PeterKietzmann, 2015-03-04:
Is there already a PR for that initialization procedure?

jfischer-phytec-iot, 2015-03-04:
@gebart @PeterKietzmann I think the idea is good.

>we could change to your proposed solution. Then we should introduce some kind of "spi initialized" flag or "spi_is_active" function to the peripheral driver so the slave device can check whether the peripheral bus is already initialized and otherwise throw an error

And such a sensor-init function should querry the parameters of sensor drivers before bus initialization.
Did I understand that correctly?

PeterKietzmann, 2015-03-04:
I didn't had this in mind. This could also be a solution but may introduce a bit more overhead then necessary. In my opinion it is up to the programmer to chose the correct parameters as this is a "natural" result of your thoughts. Like "I have this and that sensor on bus X -oh sensor A can just handle slow speeds- then I can not set the bus to high speed".
I proposed two solutions:
- Let the bus initialization inside the sensor-init function but forward the parameters from the application to guaranty there are no conflicts in bus-initialization
- Take the bus-initialization out of the sensor-drivers and do it just before sensor-initialization. In this case just check in the sensor-driver somehow if the bus is initialized (this is mainly @gebart proposal)

Now I tend to the second proposal because it's not needed to initialize the bus multiple times.

gebart, 2015-03-05:
@jfischer-phytec-iot That's an automatic solution, but that is what I meant with a convoluted solution because it is a pain to implement because of the diversity of different parameter requirements in different slave devices. It will also make the code size larger unless the optimization is performed offline (at compile time). Because of these arguments I prefer to let the application designer handle parameter definition and just use their values during bus initialization.

jfischer-phytec-iot, 2015-03-05:
>Take the bus-initialization out of the sensor-drivers and do it just before sensor-initialization. In this case just check in the sensor-driver somehow if the bus is initialized (this is mainly @gebart proposal)

Ok.

haukepetersen, 2015-03-17:
I put a little more thought into this and I think I might have a solution that I think could be feasible: how about we combine the best from both worlds:

Say we have a sensor driver that uses the SPI bus. In the device driver initialization we do then the following:

 - Check if SPI bus is already initialized
    - If not: initialize the bus as we do currently
    - If yes: skip the initialization of the SPI bus

This way, for simple use cases (one device on the bus), you can just go on simply initializing your device and this will take care of everything regarding the bus. If you have a more sophisticated setup, you can initialize the bus manually before you bring up your device drivers - and thus the drivers will skip the bus initialization. The overhead is one simple bit compare and one if condition...

Regarding the peripheral API, we need to introduce 1 small function that keeps track about the peripheral init state, maybe something like
```c
// spi.c:
static uint8_t spi_init_state = 0;

int spi_init(spi_t dev, ...) 
{
    ...
    spi_init_state |= (1 << dev);
    ...
}

// periph/spi.h:
extern uint8_t spi_init_state;

static inline int spi_is_init(spi_t dev) 
{
    return spi_init_state & (1 << dev);
}

// some device driver
int some_driver_init(..., spi_t spi, ...)
{
    ...
    if (!spi_is_init(spi)) {
        spi_init(spi, ...);
    }
    ...
}
```

gebart, 2015-03-17:
@haukepetersen I think it is better to use the strict initialization procedure.

Using the implicit initialization like you propose could potentially hide initialization mistakes because it just "happen to work" until the order of initialization of the device drivers changes.

PeterKietzmann, 2015-03-17:
@haukepetersen this is approximately what I meant some days ago. Looking from the "only one sensor on a bus" side I do like your idea. But looking from the "multiple sensors on a bus" side I agree with @gebart that there might be (hidden) initialization conflicts. Imagine you forget (or event don't know) to initialize the bus before you initialize __multiple__ sensors (this was the basic idea). Then the first sensor that is initialized will initialize the bus with its preferred parameters and we still have the problem of the smallest common parameters for all sensors on the bus. I would like to have a check in the sensor driver if the bus is initialized and otherwise return with an error. But the question is if the effort of an API change is reasonable for the benefit we get.
Is this reasonable or am I talking crap :-) ?

haukepetersen, 2015-03-17:
@PeterKietzmann: Actually I think the complexity for building a system that check on compile or runtime for minimal common parameters is too big. Also I don't like this, as this leads to parameters not clearly being defined.

>Imagine you forget (or event don't know) to initialize the bus before you initialize multiple sensors (this was the basic idea)

(i) this behavior is/needs to be clrearly stated in the documentation and (ii) I still have the strong opinion, that we can expect some extend of knowledge by the users/developers so that they are actually able to figure out the best parameter set for their setup and initialize their peripherals accordingly!

PeterKietzmann, 2015-03-17:
Maybe I didn't  stated my point that well.

- I am also against a system that checks parameters as this is too complex
- Regarding (i) is agree that this must be documented well
- Regarding (ii) I also agree that we can expect the knowledge of the best parameters by the users

I am pro for doing bus initializations outside the driver. I am sceptic about your proposed solution where the bus is initialized inside the sensor driver when the bus is not initialized, as this may cause conflicts again. I could imagine a solution where you just check in the sensor driver  __if__ the bus is initialized and if not return with an error. But this would also require an API change and may be discussed as the effort could exceed the benefit of this.

gebart, 2015-03-17:
@haukepetersen 
> (i) this behavior is/needs to be clrearly stated in the documentation and (ii) I still have the strong opinion, that we can expect some extend of knowledge by the users/developers so that they are [...]

I have the exact same opinion about the expectation in knowledge of the users, which is why I think we should let the users handle initializing.

@PeterKietzmann I agree with your arguments. Signalling an error when the bus has not been initialized before initializing the device driver would be helpful for developers, but requires an API change since many initialization functions return `void` IIRC.

gebart, 2015-05-21:
@haukepetersen  What is your stance on this issue today? 

I strongly believe in moving bus initialization outside the device drivers to allow for more precise control over bus parameters. I will run into problems in the future when I start using the on board memories on the Mulle together with the radio and accelerometer if I can not force the bus to go into the lowest common communication speed. (The memories can be driven at around 80 MHz or so)

I can perform the tedious work of moving the init calls outside the drivers on the code that is in-tree (`drivers`, `tests/driver_*`), but I can't modify applications for other people which may be outside of the open source project.

PeterKietzmann, 2015-05-22:
I am still in favor of taking the initialization outside the driver. After reading our past comments I still think that the following could be a good solution, even if I'm not totally convinced that it is a "pretty" and efficient solution, but more feels like an alternative way of "getters" like e.g. in netdev.:
When taking the bus initialisation procedure outside a sensor driver, we could extend the bus driver APIs with a function that returns the speed a bus is initialised with. Of course an `UNDEF` parameter says that the bus isn't even initialised. A sensor driver checks this value and returns an error if the bus speed does not fit its requirements.

haukepetersen, 2015-05-22:
Hej @gebart and @PeterKietzmann, 
as you know, I am not at all convinced of taking the peripheral initialization out of the device drivers. My main reasons so far were (i) simplicity of usage, (ii) a driver knows best about it's needed parameters, and (iii) setups with contradicting parameters for a shared device were the absolute minority.

Now as we learned, (iii) does not hold anymore (and I guess we should have anticipated this...). (i) and (ii) do still hold. I think forcing a user to initialize the peripherals manually introduces quite some overhead/complexity for many use-cases. So as stated above, I would still opt for a dual solution, where you can (if you choose to) initialize your peripherals manually, but by default this is still done inside the device drivers.

PeterKietzmann, 2015-05-22:
I think one could integrate this default configuration in my proposal above. Instead of just returning an error if the bus speed does not fit the requirements, one could also initialize the device with its default value. The outside initialization should have priority to leverage conflicts we want to solve within this issue :-).
Just to explain my general point of view: I also like to keep things as simple as possible and leave the detailed implementation concerns to the user.  But as I can't acknowledge (iii) (maybe because of missing experience outside the OS world,) I just see @gebart 's problems which seem reasonable to me.

daniel-k, 2015-09-02:
Ping. I just stumbled over peripheral intialization when implementing a device driver and @PeterKietzmann pointed me to this issue. I don't have a too strong opinion but I tend to like moving the initialization out of the driver a bit more.

IldarValiev, 2015-11-03:
In case of SPI, there can be implemented "SPI profiles" that can be used during communication with specific sensor. Within SPI transaction starting procedure, current bus profile can be checked and compared with desired. If they are equal - just start transaction, else - change it to desired and start transcation. If there are only one profile for all sensors it can be defined as, for example,
`#define SPI_PROFILE_DEFAULT spi_profile_1`

And functions will look like
`int spi_transfer_byte(spi_t dev, spi_profile_t profile, char out, char *in);`

Pros:
- SPI initialization can be left inside of a driver. The initialization procedure will use most appropriate settings for this sensor;
- Possibility to use sensors with different requirements to speed, frame structure, etc;

Cons:
- The functions become more complicated;
- API change;

PeterKietzmann, 2015-11-03:
@IldarValiev thanks for your proposal and triggering that topic :-) ! Do I understand it correctly that you would change the rate for each transfer (per device), in a worst-case scenario where multiple sensors with different requirements are connected to one bus? My personal impression is that we'd rarely need/use such a feature in RIOT. And therefore it seems to me like the overhead does not justify the benefit. A solution where the programmer manually defines the "highest possible but smallest needed" baudrate still seems reasonable to me. But let's wait for the opinions of the other guys.

haukepetersen, 2015-11-03:
The approach with SPI profiles sounds interesting, though I fear we can not use it: it completely breaks any real-time properties! A SPI transfer will take a non-deterministic amount of time (because there might, but might not, be a re-initialization/change of parameters), so you can not design drivers which need to follow strict timings...

IldarValiev, 2015-11-04:
@PeterKietzmann, thank you :) If multiple sensors _can_ work with one SPI profile then it comes to one profile which can be set once and never change again. My proposal is for cases when several sensors _cannot_ work with the same settings. For example, one sensor supports only low speed communication, while data from the other should be read on high speed to remain actual or because of big data flow. I understand that such cases are very rare and efforts may not cost results, also it may be out of scope of RIOT-OS, but this is just proposal :)

@haukepetersen, what real-time properties does this feature break? You don't use if-statements? As far as I know, to change configuration of an SPI module in any CPU only several regiesters should be altered, no waiting, no hanging anywhere. Time needed to change a register is quite determined, so you can calculate time needed to all change process. Also, in `int spi_transfer_byte(spi_t dev, char out, char *in)` I found polling _while_ which breaks your real-time properties as well :)

P.S. In my opinion this feature is not very important for IoT on this level of development but, as for me, RIOT-OS tries to include everything in it, instead of focusing on most important aspects of OS for IoT, that's why I suggested this feature.

PeterKietzmann, 2015-11-04:
@IldarValiev now I got it. Still I fear that such kind of implementation is out of scope. Until now, I've never seen the situation like described by you. Do you personally have a use-case?
I don't want to scare you away (in no way!), either I'm really happy about new people putting their ideas and thoughts to RIOT and sharing their experiences with us. But I don't think that I will implement your proposal. IMO it is still too much overhead for just covering a "corner-case" scenario. 
> instead of focusing on most important aspects of OS for IoT

What are the most important aspects in your opinion? (I'm just curious :-) )

gebart, 2015-11-04:
@PeterKietzmann @IldarValiev 
I have a similar use case where there are two devices on the same bus which do not operate with the same clock polarity and phase. On Mulle we implemented it as two virtual `SPI_x` buses, but both sharing the same hardware bus.
https://github.com/RIOT-OS/RIOT/blob/master/boards/mulle/include/periph_conf.h#L364

The K60 can have two different configurations set at the same time in hardware and when initiating a transfer we also need to tell the hardware which configuration to use (CTAS).

The current solution on mulle with two virtual buses is just a workaround for the limitations of the current SPI API, I would very much like to see some more flexibility in the API, but it has to be done in a clever way to still allow the device drivers to be platform independent. #3216 is some work in progress started by @haukepetersen on his quest to improve the low level driver APIs, but like he wrote yesterday he might not have time to work on it for the time being.

@IldarValiev Do you have the time and desire to work on improving the SPI interface? I do not have time to work on it myself right now but I can help with reviewing and testing.

See also #2116 which is discussing possible changes to the SPI interface as well.

IldarValiev, 2015-11-08:
> What are the most important aspects in your opinion? (I'm just curious :-) )
@PeterKietzmann, anything that will bring benefits to devices for _Internet of Things_. As I understand IoT, most devices are located in hard-to-reach places, so in case of malfunction there should be possibility to reset such device in automatic (watchdog) or manual remote mode (mmm.. hypervisor?). Another must-have feature in my point of view is remote firmware update mechanism.
Such features should be a part of the core of OS for IoT.
We can discuss all this, but not in this topic. If you are interested in the continuation of this discussion, email me. :)

@gebart I have desire to work on this, but I am not sure if I have time for this. I'll answer you later when it becomes more clear.

lebrush, 2016-08-01:
Only for completion: It's also being tackled in #4926 for i2c. Current approach is to define a fixed speed for the bus in the `periph_conf.h` file.

miri64, 2016-10-17:
So I guess #4780 and #4926 are fixing this issue?

PeterKietzmann, 2016-10-29:
@miri64 as far as I'm currently involved I think yes. However, unfortunately both PRs were postponed for next release. @haukepetersen can you give a comment here and move the Milestone label then?

miri64, 2016-10-29:
Please don't move the milestone but move the issue to the appropriate  "known issues" column then.

haukepetersen, 2017-01-17:
sounds good.

closed 2019-09-10 by stale[bot]