RIOT-OS/RIOT #2531: core: disableIRQ/restoreIRQ inefficient?

kaspar030, 2015-03-04:
I've been thinking whether we might save some cycles by avoiding unnecessary enable/restoreIRQ calls.

Right now we blindly disable them, and blindly restore them after we don't use them anymore.

Almost everywhere a pair of disable/restore is used to enter/leave sections which should not be interrupted.

We could use a guard variable (e.g., irq_disabled), which gets incremented when entering a critical section, and decremented when leaving, and only do the actual disabling/restoring when that variable changes from or to 0.

As the check for disabling would need an atomic operation, blindly disabling might be faster than a check, depending on an architectures.

Thoughts?

gebart, 2015-03-09:
I have not studied the assembly code from such an implementation but I think it is unlikely that an atomic counter would be faster than the disableIRQ/restoreIRQ pair. The memory gains would also be minimal since the IRQ mask is a few bytes and will likely remain in a register during the whole critical section.

On the other hand, maybe we should look for nested critical sections instead and reconsider if they really are critical sections or if they can be allowed to be interrupted.

kaspar030, 2015-03-10:
I tried the following (best case for the "optimization"):

<pre>
int _in_critical = 0;
static inline void enter_crit(void) {
    disableIRQ();
    _in_critical++;
}

static inline void leave_crit(void) {
    if (!_in_critical--) {
        enableIRQ();
    }   
}

int main(void)
{
    unsigned state;
    printf("%u\n", hwtimer_now());
    for (unsigned long N = 100000; N > 0; N--) {
//        state=disableIRQ();
          enter_crit();    
    }   
    for (unsigned long N = 100000; N > 0; N--) {
//        restoreIRQ(state);
          leave_crit();    
    }   

    printf("%u\n", hwtimer_now());
    return 0;
}
</pre>

This is already 10% faster without the optimization (on iot-lab_M3), so I'll close this.

+1 for analyzing all critical sections! ;)

closed 2015-03-10 by kaspar030