RIOT-OS/RIOT #2550: stm32f4: Hex file from make doesn't run

jhollister, 2015-03-07:
I'm trying to run some of the examples on my stm32f4-discovery board and was having issues flashing the compiled hex files onto the board. I found that changing the objcopy OFLAGS in the board's Makefile.include from `-O ihex` to `-O binary` fixed this. 

Can anyone else confirm that this problem and solution aren't unique to my board? I had a look at the other Makefile.include and it seems pretty split between `-O binary` and `-O ihex` so maybe they are all supposed to be binary?

thomaseichinger, 2015-03-07:
@jhollister Thanks for bringing this up. Will test this on Monday as I don't have the hardware with me right now. I'm wondering as with the flashing tool (OpenOCD) should be agnostic to the given file format.

jhollister, 2015-03-07:
I forgot to mention that I'm using [stlink](https://github.com/texane/stlink) st-flash and not OpenOCD. I haven't tried with OpenOCD yet.

thomaseichinger, 2015-03-07:
With bea0171754adbd9808b3a93295a5fbbdc679866c we moved from stlink to OpenOCD with which the file format was changed. We could consider changing it to `binary` for compatibility though.

jhollister, 2015-03-08:
Ok, that makes sense. What was the reason for the switch? I haven't worked with openocd before but I will get it set up and perhaps also update the documentation [here](https://github.com/RIOT-OS/RIOT/wiki/Getting-started-with-STM32F%5B0%7C3%7C4%5Ddiscovery-boards) for use with openocd.

gebart, 2015-03-08:
Isn't `-O binary` completely naked without any offset information etc? i.e. if there are multiple memory segments in the file at different address ranges the file will be large enough to cover both? (albeit with file holes it does not take up any disk space, but is still a hassle to transfer between systems)

Intel hex contains offset addresses for the data so a hex file can cover multiple address ranges and still only be somewhat compact (as compact as an ascii representation of hexadecimal numbers can be)

gebart, 2015-03-08:
also, for flashing a raw binary you'll need to specify the starting offset, all the other possible formats (ELF, ihex etc) carry this information inside the file.

OlegHahm, 2015-03-11:
Couldn't find any reliable information about the format of `-O binary`, but I would stay with `-O ihex` to be on the safe side. Can we close this issue then?

jhollister, 2015-03-12:
I'm fine with closing it, although I think the wiki should be updated for instructions on OpenOCD instead of stlink. I'll try to do it if I have time but it's finals week for me right now so I don't have much free time. Should I be the one closing it since I opened it?

LudwigOrtmann, 2015-03-12:
@jhollister regarding the wiki - it already contains instructions about openocd.

closed 2015-03-12 by LudwigOrtmann