RIOT-OS/RIOT #2557: info-build is broken for unittests

noshky, 2015-03-08:
When calling

``make info-build``

from within the ``RIOT/tests/unittests`` directory, ``make`` breaks after processing the ``CFLAGS`` variable.
I've tracked this down to happen in ``Makefile.buildtests:168``, where the entries in ``CFLAGS`` are separated by ``\n\t`` for screen presentation. The ``CFLAGS`` contains test-strings such as 

``-DTEST_STRING12="50U'HLKC3_ft"``

(defined in ``tests/unittests/Makefile:18``). This string (and others) contains the character ``'``, which breaks the quoting for ``make``.

I didn't find a solution so far for neither escaping the quotes in the definition nor handling the string better in the ``patsubst`` call.

authmillenon, 2015-03-08:
The simplest solution might be to remove the quote char from the test string ;-).

OlegHahm, 2015-03-09:
Maybe https://github.com/OlegHahm/RIOT/commit/b397121cffc9b4835eff9e54b4db1c5d02813480 helps.

noshky, 2015-03-11:
No, OlegHahm@b397121 unfortunately doesn't help. If the quote char isn't needed, maybe we can just remove it? When I remove the three single quotes, ``make info-build`` runs fine, and
``cd tests/unittests``
``make clean all term``
gives me ``OK (317 tests)``.

OlegHahm, 2015-03-11:
I would prefer a solution where we can run unittests with arbitrary strings.

phiros, 2015-03-15:
My first idea would be to remove the ```'``` (BTW: I think this would be the most pragmatic solution to the problem).
However, if we really want to fix this we might need to pass arrays of data (in which chars are encoded as hex values) to tests instead of strings. This would require a little bit of test code to be changed.

LudwigOrtmann, 2015-03-15:
I would just move those strings into a header instead of passing them around in command line parameters.

authmillenon, 2015-03-15:
The idea of passing them around as command-line parameters is to change them at compile time if needed, without changing any files.

LudwigOrtmann, 2015-03-16:
Is that really important? Did you ever do this (after the PR was merged) and if so: why?
I think if the strings need to be changed for testing locally it means they should be changed permanently because the tests are insufficient.

OlegHahm, 2015-03-18:
I agree with @LudwigOrtmann.

phiros, 2015-03-20:
I agree with @LudwigOrtmann and @OlegHahm.
#2663 should fix this problem.

phiros, 2015-03-22:
merged #2663 -> fixed

closed 2015-03-22 by phiros