RIOT-OS/RIOT #2611: boards/Makefile.features: one feature per row

haukepetersen, 2015-03-16:
Most boards list their feature in one or two rows, e.g.:
```
FEATURES_PROVIDED += periph_uart periph_gpio periph_spi periph_i2c periph_pwm periph_random \
                     periph_adc periph_dac periph_cpuid
```
This leads to conflicts when there are two PR's open simultaneously and both add something to this line. I would actually prefer to have one line per feature, e.g.:
```
FEATURES_PROVIDED += periph_uart
FEATURES_PROVIDED += periph_spi
FEATURES_PROVIDED += periph_i2c
FEATURES_PROVIDED += periph_pwm
...
```
This way there won't be any merge conflicts anymore.

If everyone agrees, I will include a TODO list with boards that need to be touched...

BigDaddyD, 2015-03-16:
It seems like an easy fix that with both minimize conflcits and improve readability

kushalsingh007, 2015-03-16:
So what boards do you suggest then @haukepetersen ?

LudwigOrtmann, 2015-03-16:
all of them

haukepetersen, 2015-03-17:
list added, see above

Darredevil, 2015-03-22:
I wanted to tackle this but I noticed PR #2641 already does that but isn't merged,even Travis CI says everything is ok. Any reason for that?

LudwigOrtmann, 2015-03-23:
Probably it was lost in the flood of PRs or the assigned maintainer didn't have time to look into it.

PR #2641 (2015-03-23): boards/Makefile.features: one feature per row
Boards list their features one per row compared to all their features on one or two rows. This fixes #2611.
closed 2015-03-23 by PeterKietzmann