RIOT-OS/RIOT #2623: Repair header file include guards

haukepetersen, 2015-03-17:
As we were pointed out a while ago, defines that start with underscores are not allowed by the C standard (I forgot the PR number where we discussed this -> could somebody help here?). 

So the following style
```c
#ifndef __MSG_H_
#define __MSG_H_

...

#endif /* __MSG_H_ */
```
would need to be changed to something like
```c
#ifndef MSG_H_
#define MSG_H_
...
#endif /* MSG_H_ */
```

Git grep tells me there are a lot of files that need to be adjusted. So if anyone is taking this on, please to it in small, easily reviewable and logical grouped chunks...

gebart, 2015-03-17:
Use the following command to find the offending include guards:

```sh
git grep -E 'ifndef\s*__'
```

BigDaddyD, 2015-03-18:
So writing a script for this change is out of the question?

gebart, 2015-03-18:
You may choose whatever method you like, a script is fine as long as it does the right thing. We will still need to do a manual code review for the change if/when you open a PR.

kushalsingh007, 2015-03-18:
working on this one :)

shrenikjain38, 2015-03-18:
Do we have to edit the defines starting with single underscores as well?
And by defines is it meant that it has to be done for #define and #ifdef as well?

gebart, 2015-03-18:
@shrenikjain38 Changing `_FOO` is a good idea. Identifiers beginning in single underscore followed by a upper case letter is also reserved.

btw, your scripted commit https://github.com/shrenikjain38/RIOT/commit/fd40ba441aa3ac6e1223d14ba275b5ee15d5e5eb breaks the C++ checks (don't modify `#ifdef __cplusplus`)

The `#ifdef HEADER_FILE_NAME_H_` and `#define HEADER_FILE_NAME_H_` are include guards that protect against multiple inclusion of the same file, both `#define` and `#ifdef` must be changed together.
Also, don't forget that there is usually a comment at the `#endif` containing the same symbol name as well which should be updated for consistency. e.g. `#endif /* HEADER_FILE_NAME_H_ */`

gebart, 2015-03-18:
another hint for anyone preparing pull requests for this issue: Read the description at the top, we won't be able to merge the changes if you don't split them into smaller logical groups.

>So if anyone is taking this on, please to it in small, easily reviewable and logical grouped chunks...

shrenikjain38, 2015-03-18:
>Changing _FOO is a good idea. Identifiers beginning in single underscore followed by a upper case letter is also reserved.
Can you please elaborate it a bit more if the identifiers beginning with a single underscore are to be changed and if they are, whether those followed by an upper case letter after an underscore to be modified or not?
>Also, don't forget that there is usually a comment at the #endif containing the same symbol name as well which should be updated for consistency. e.g. #endif /* HEADER_FILE_NAME_H_ */
This issue has already been taken into consideration.

Is it okay if the changes are made in a specific directory to divide the whole procedure in easily reviewable and logically grouped chunks?

Darredevil, 2015-03-22:
There are still files with this issue, in /drivers for example, is it ok if I patch them even if this PR has been closed? (will not create an avalanche of new PRs this time :)

authmillenon, 2015-03-23:
The issue was automatically closed by github, since there was the message "fixes 2623" in your PRs somewhere. I anticipated that when I merged the PRs but forgot to reopen the issue. Sorry for that.

PeterKietzmann, 2015-03-30:
How do we proceed with trailing underscores? I found some guards with and without them.

authmillenon, 2015-03-31:
There are still occurrences in master.

BigDaddyD, 2015-04-18:
I am currently working on fixing all headers under the boards directory. I managed to use git grep and sed to find and fix all ifndefs and defines. I was wondering if that can be used to fix the comments as well

PeterKietzmann, 2015-07-03:
@BigDaddyD what's the status? Are you still working on it?

miri64, 2016-10-17:
There are still some of these macros left. In sys in particular.

zhuoshuguo, 2016-10-17:
@miri64 Will find time to handle.

closed 2017-01-20 by miri64

miri64, 2017-01-20:
Some are still left by git's non-concurrency ;-):

```
[mlenders@mk RIOT]<3 git grep "H_$"
boards/nucleo32-f303/include/board.h:#ifndef BOARD_H_
boards/nucleo32-f303/include/board.h:#define BOARD_H_
boards/nucleo32-f303/include/periph_conf.h:#ifndef PERIPH_CONF_H_
boards/nucleo32-f303/include/periph_conf.h:#define PERIPH_CONF_H_
cpu/k60/include/MK60D10.h:#define MK60D10_H_
cpu/k64f/include/MK64F12.h:#define MK64F12_H_
cpu/kw2x/include/MKW22D5.h:#define MK22D5_H_
cpu/native/include/netdev2_tap_params.h:#ifndef NETDEV2_TAP_PARAMS_H_
cpu/native/include/netdev2_tap_params.h:#define NETDEV2_TAP_PARAMS_H_
examples/ccn-lite-relay/Makefile:CFLAGS += -DCCNL_UAPI_H_
Binary file pkg/emb6/patches/0002-Rename-colliding-files-and-functions.patch matches
Binary file pkg/oonf_api/patches/0001-add-RIOT-support.patch matches
Binary file pkg/oonf_api/patches/0009-Add-missing-include.patch matches
sys/include/uuid.h:#ifndef UUID_H_
sys/include/uuid.h:#define UUID_H_
```

PR #2675 (2015-03-22): unittests/netdev_dummy: fixed header file include guard
fixes #2623
commit 49c2f5c5 (2015-03-22): unittests/netdev_dummy: fixed header file include guard

fixed #2623
PR #2682 (2015-03-22): unittests/tests-pktbuf: fixed header file include guard
fixes #2623
commit 7c971366 (2015-03-22): unittests/tests-pktbuf: fixed header file include guard

fixes #2623