RIOT-OS/RIOT #2634: posix/pthread: unknown type name `clockid_t` on OS X

thomaseichinger, 2015-03-18:
Compilation fails for all `tests/pthread*` tests with the following message
```bash
In file included from RIOT/sys/posix/pthread/include/pthread.h:41:
RIOT/sys/posix/pthread/include/pthread_cond.h:87:63: error:
      unknown type name 'clockid_t'; did you mean 'clock_t'?
int pthread_condattr_getclock(const pthread_condattr_t *attr, clockid_t *clock_id);
                                                              ^~~~~~~~~
                                                              clock_t
/usr/include/sys/_types/_clock_t.h:30:33: note: 'clock_t' declared here
typedef __darwin_clock_t        clock_t;
```

Compiler version
```bash
% gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.0 (clang-600.0.57) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.1.0
Thread model: posix
```

OlegHahm, 2015-03-18:
Duplicate? https://github.com/RIOT-OS/RIOT/issues/2360

thomaseichinger, 2015-03-18:
yup, overlooked it.

closed 2015-03-18 by thomaseichinger