RIOT-OS/RIOT #2642: rpl: root cannot send to nodes

cgundogan, 2015-03-19:
Using the rpl_udp example, a root cannot send a message to another node via `send id test`. Looking at wireshark, the 802.15.4 destination is set to broadcast `0xffff` instead of the node's id and thus results in dropping the packet.
The change was introduced in commit: 70cf054b5e1cb539ce17f0cdf49be55fb133e10a.

gebart, 2015-03-19:
I don't understand this, it worked for me on both native and mulle when I tested it before merging. What board are you using? 

The send command needs some improvements though. I would like to be able to type an IP address instead of only a 16 bit short 802.15.4 address or broadcast.

cgundogan, 2015-03-19:
@gebart I tested with native. Communication works from node to node, but as soon as I use the `send` command from root to node, it breaks like described above. I did not look much into it, maybe later in the evening or tomorrow

BytesGalore, 2015-03-30:
@cgundogan the problem seems to be only native related. 
I tested sending on IoT-lab_M3 nodes with success (delivered the packet in either direction). 
On native however sending from root to node seems to be disrupted. Sending from node to root works for me.

OlegHahm, 2015-04-20:
Fixed with #2831.

closed 2015-04-20 by OlegHahm