RIOT-OS/RIOT #2648: stm32 I2C START condition issue

yerpj, 2015-03-19:
Hi all,

I am currently porting RIOT on a stm32f2 (f217) and I am experiencing issues with the i2C.
I tried on both stm32f4discovery and on our own board. Result is the same.
_start() is executing (periph/i2c.c) but after writing the dev->DR register, an interrupt occurs. Debug mode tells "I2C ERROR OCCURED", "status: 00000400", "AF", which means "AF: Acknowledge failure". But, having checked with the oscilloscope, the address is not transmitted, only the START condition is sent, before being interrupted by the isr. 
Here is the _start() function:

static void _start(I2C_TypeDef *dev, uint8_t address, uint8_t rw_flag)
{
    /* wait for device to be ready */
    DEBUG("Wait for device to be ready\n");

    while (dev->SR2 & I2C_SR2_BUSY);

    /* generate start condition */
    DEBUG("Generate start condition\n");
    dev->CR1 |= I2C_CR1_START;
    DEBUG("Wait for SB flag to be set\n");

    while (!(dev->SR1 & I2C_SR1_SB));

    /* send address and read/write flag */
    DEBUG("Send address\n");
    dev->DR = (address << 1) | rw_flag;
    /* clear ADDR flag by reading first SR1 and then SR2 */
    DEBUG("Wait for ADDR flag to be set\n");

    while (!(dev->SR1 & I2C_SR1_ADDR));
}

Did someone already encountered this issue?

[image]

PeterKietzmann, 2015-03-19:
Hi @yerpj. Did you measure with an "open bus" ? If so, can you try it again with an i2c-slave device connected? Also make shure you have appropriate pull-up strength.

yerpj, 2015-03-19:
Hi @PeterKietzmann : Thanks for your reply. I have some slaves on the bus. I have external pull-ups of 10k (the time-scale on the picture is 250ns per square; the "slow" low to high transition is actually fast enough). 
The i2c error occurs before any slave answers, because the address is not sent. Furthermore, the START condition is wrong, as the SDA should be asserted before the SCL. 
Did anyone manage to get the I2C bus working on STM32? I don't know if I can trust the stm32f4discovery/periph/i2c.c driver...  
Thank you!

PeterKietzmann, 2015-03-19:
Yes I ran it anytime. Which speed do you use? I won't accomplish to test it now but then I'll do so tomorrow.

yerpj, 2015-03-19:
I am using the I2C_SPEED_NORMAL, maybe the CCR register is not correctly inited, as it depends on APB clock, which is defined in my periph_conf.h file. I will check that point soon... The weird thing is, when I try the same code on stm32f4discovery, the _start() function has the same behaviour and the I2C issue still happens....

PeterKietzmann, 2015-03-19:
@yerpj I gave it a quick try and it worked. My setup is a stm32f4discovery board and the srf02 ultrasonic senso without external pull-ups. The test in "RIOT/tests/driver_srf02" worked and the start condition looks ok as you can see :/ ...
[image]

PeterKietzmann, 2015-03-19:
Ups. Please note that SDA and SCL are wrongly named here

yerpj, 2015-03-19:
This seems to work perfectly on STM32F4DISCOVERY. 

I found the issue, thanks to your question about the i2c speed! 
I did not adapt the I2C_APBCLK in accordance with my CPU speed, which is 120MHz (max) for the STM32F217. Therefore, the config should be 
#define I2C_APBCLK          (30000000U)   //CPU_SPEED_MAX/APB_PRESCALER=120M/4=30M
The I2C is not perfectly working. I will study this I2C driver in details, as it has to provide a timeout in case of the i2c slave doesn't answer to its address. 

Thank you for your support! I will do a PR of the stm32f2 port shortly, but in the meantime I have to learn about how git and pull-requests are working:-)

PeterKietzmann, 2015-03-20:
@yerpj nice to hear :-) ! Regarding the PR and git stuff these wiki pages may help you: [Contributing to RIOT](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT) and [Git cheatsheet](https://github.com/RIOT-OS/RIOT/wiki/Git-cheatsheet). Otherwise feel free to ask on the mailing list or at least in you PR when it's there.

PeterKietzmann, 2015-03-20:
Can we close this issue, is it solved?

yerpj, 2015-03-23:
Yes it is. Thanks for the support!

PeterKietzmann, 2015-03-23:
Nice

closed 2015-03-23 by PeterKietzmann