RIOT-OS/RIOT #2657: ubjson: Missing types compiler errors on OSX

jhollister, 2015-03-20:
Compiler gives the following errors for `unittests/ubjson`:
```
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/jhollister/dev/RIOT/sys/ubjson
In file included from /Users/jhollister/dev/RIOT/sys/ubjson/ubjson-read.c:29:
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:211:19: warning: type specifier missing, defaults to 'int'
      [-Wimplicit-int]
typedef ssize_t (*ubjson_read_t)(ubjson_cookie_t *__restrict cookie, void *buf, size_t max_len);
                 ~^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:211:9: warning: type specifier missing, defaults to 'int'
      [-Wimplicit-int]
typedef ssize_t (*ubjson_read_t)(ubjson_cookie_t *__restrict cookie, void *buf, size_t max_len);
~~~~~~~ ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:211:17: error: function cannot return function type 'int
      (ubjson_cookie_t *restrict, void *, size_t)'
typedef ssize_t (*ubjson_read_t)(ubjson_cookie_t *__restrict cookie, void *buf, size_t max_len);
                ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:243:34: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
typedef ssize_t (*ubjson_write_t)(ubjson_cookie_t *__restrict cookie, const void *buf, size_t len);
                                 ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:254:9: error: unknown type name 'ubjson_read_t'; did you mean
      'ubjson_type_t'?
        ubjson_read_t read;   /**< read function */
        ^~~~~~~~~~~~~
        ubjson_type_t
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:166:3: note: 'ubjson_type_t' declared here
} ubjson_type_t;
  ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:301:57: error: unknown type name 'ubjson_read_t'; did you mean
      'ubjson_type_t'?
                                                        ubjson_read_t read,
                                                        ^~~~~~~~~~~~~
                                                        ubjson_type_t
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:166:3: note: 'ubjson_type_t' declared here
} ubjson_type_t;
  ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:329:23: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_get_i32(ubjson_cookie_t *__restrict cookie, ssize_t content, int32_t *dest);
                      ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:338:23: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_get_i64(ubjson_cookie_t *__restrict cookie, ssize_t content, int64_t *dest);
                      ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:349:26: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_get_string(ubjson_cookie_t *__restrict cookie, ssize_t content, void *dest);
                         ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:359:38: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
static inline ssize_t ubjson_get_bool(ubjson_cookie_t *__restrict cookie, ssize_t content, bool *dest)
                                     ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:373:39: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
static inline ssize_t ubjson_get_float(ubjson_cookie_t *__restrict cookie, ssize_t content, float *dest)
                                      ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:392:40: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
static inline ssize_t ubjson_get_double(ubjson_cookie_t *__restrict cookie, ssize_t content, double *dest)
                                       ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:452:26: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_null(ubjson_cookie_t *__restrict cookie);
                         ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:459:26: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_noop(ubjson_cookie_t *__restrict cookie);
                         ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:467:26: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_bool(ubjson_cookie_t *__restrict cookie, bool value);
                         ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:475:25: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_i32(ubjson_cookie_t *__restrict cookie, int32_t value);
                        ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:484:25: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_i64(ubjson_cookie_t *__restrict cookie, int64_t value);
                        ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:492:27: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_float(ubjson_cookie_t *__restrict cookie, float value);
                          ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:500:28: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_double(ubjson_cookie_t *__restrict cookie, double value);
                           ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:509:28: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_write_string(ubjson_cookie_t *__restrict cookie, const void *value, size_t len);
                           ^
/Users/jhollister/dev/RIOT/sys/include/ubjson.h:518:26: error: function cannot return function type 'ssize_t'
      (aka 'int (int *)')
ssize_t ubjson_open_array(ubjson_cookie_t *__restrict cookie);
                         ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
2 warnings and 20 errors generated.
```

Seems to be missing a header file and if you look at `sys/include/ubjson.h:41` this seems to be the issue:
```
#elif !defined(CPU_NATIVE)
#   include <sys/types.h>
#endif
```

Tried changing to:
```
#elif !defined(CPU_NATIVE) || defined(__MACH__)
#   include <sys/types.h>
#endif
```
Which would let me compile (needs the malloc.h fix from PR #2590) but running the actual unit test fails with the output:
```
$: make clean tests-ubjson term
__TEXT  __DATA  __OBJC  others  dec     hex
49152   1052672 0       24576   1126400 113000
/Users/jhollister/dev/RIOT/tests/unittests/bin/native/unittests.elf
RIOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

kernel_init(): This is RIOT! (Version: 2014.12-715-gc5da8-jhollister-mbp-ubjson_fix_osx)
kernel_init(): jumping into first task...
..% 
```

OlegHahm, 2015-09-03:
@thomaseichinger, @Kijewski, wanna take a look at this?

thomaseichinger, 2015-09-03:
@OlegHahm should be fixed with https://github.com/RIOT-OS/RIOT/pull/3768

thomaseichinger, 2015-10-01:
Fixed by #3768.

closed 2015-10-01 by thomaseichinger