RIOT-OS/RIOT #2660: question in arm compile

sirtian, 2015-03-20:
When I compile default example by "make BOARD=stmf32f4discovery"
usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: /home/RIOT-OS/RIOT/examples/default/bin/stm32f4discovery/default.elf uses VFP register arguments, /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-errno.o) does not
/usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-errno.o)
Is it because of the compile itself error?which version are you using to compile arm family application? can any one help me?I appreciate it very much

gebart, 2015-03-20:
Your installed newlib version is built for soft float, the RIOT application is built for hard float. If you don't want to change your newlib you can try adding `-mfloat-abi=softfp` to your CFLAGS and do a `make clean`

sirtian, 2015-03-21:
@gebart , thanks for your answer,after I added -mfloat-abi=softfp both in  Makefile.cflags and  boards/stm32f4discovery/Makefile.include, I still get the following error messages:
/usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: /home/RIOT-OS/RIOT/examples/default/bin/stm32f4discovery/default.elf uses VFP register arguments, /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-errno.o) does not
/usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-errno.o)
/usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: /home/RIOT-OS/RIOT/examples/default/bin/stm32f4discovery/default.elf uses VFP register arguments, /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-impure.o) does not
/usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-impure.o)
......
/usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/thumb2/libgcc.a(_udivdi3.o)
collect2: error: ld returned 1 exit status
make: *** [all] Error 1
Is it because of my wrong operate? The arm compiler version is: arm-none-eabi-gcc --version
arm-none-eabi-gcc (Sourcery CodeBench Lite 2014.05-28) 4.8.3 20140320 (prerelease)
Copyright (C) 2013 Free Software Foundation, Inc. 
Here is my changes:
+++ b/Makefile.cflags
@@ -50,6 +50,7 @@ endif
 
 # Forbid common symbols to prevent accidental aliasing.
 CFLAGS += -fno-common
+CFLAGS += -mfloat-abi=softfp
+++ b/boards/stm32f4discovery/Makefile.include
@@ -32,7 +32,7 @@ export RESET = $(RIOTBASE)/dist/tools/openocd/openocd.sh
 
 # define build specific options
 CPU_USAGE = -mcpu=cortex-m4
-FPU_USAGE = -mfloat-abi=hard -mfpu=fpv4-sp-d16
+FPU_USAGE = -mfloat-abi=softfp -mfpu=fpv4-sp-d16

Could you help me to find what is wrong with this arm compile problem?

gebart, 2015-03-21:
Try different combinations of the float configuration flags `-msoft-float`, `-mfloat-abi={soft,softfp}`

sirtian, 2015-04-12:
@OlegHahm please help me to find what's wrong?

gebart, 2015-04-12:
What have you tried so far?

OlegHahm, 2015-04-12:
Is there a reason not to use the recommended ARM toolchain from launchpad?

gebart, 2015-05-09:
Since the original poster has not responded for a long time and the toolchain is not the recommended one I will now close this issue.

closed 2015-05-09 by gebart

rykerDial, 2016-11-30:
I ran into the same issue trying to compile a project for the Teensy 3.6. with arm-none-eabi 4.8.4. The issue was that gcc was linking to _libarm_cortexM4l_math.a_, which is the software floating point version of the library. The solution was to force it to link to _libarm_cortexM4lf_math.a_, the hardware floating point version

nilesh-xebrium, 2018-03-07:
Hello rykerDial,
how to force it link to libarm_cortexM4lf_math.a. 
thanks

zeran86, 2018-10-04:
> I ran into the same issue trying to compile a project for the Teensy 3.6. with arm-none-eabi 4.8.4. The issue was that gcc was linking to _libarm_cortexM4l_math.a_, which is the software floating point version of the library. The solution was to force it to link to _libarm_cortexM4lf_math.a_, the hardware floating point version, instead.

Can you explain pls how to do this? Thanks!

rykerDial, 2018-10-04:
Hmm, well it's been a while. If I remember correctly my solution was pretty inelegant. I simply went into the arm-none-eabi folder and renamed libarm_cortexM4lf_math.a to libarm_cortexM4l_math.a.

swd543, 2018-10-07:
@rykerDial that indeed is inelegant and would still compile using software float. Have you figured out a way to link the hard float library while compiling?

swd543, 2018-10-07:
I am facing the same issue as @rykerDial and I confirmed this by changing the makefile to using soft float abi. With the soft float abi it compiles well but using the hard float abi gives an error 
```error: build/STMBuga.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(_exit.o) does not```

vrushalibhokare, 2018-10-23:
I am facing same issue trying to compile project for mbed cli with compiler version 5 . 
[ERROR] c:/users/admin/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: error: ./STM32_Audio/Addons/PDM/Lib\libPDMFilter_CM4_GCC.a(pdm_filter.o) uses VFP register arguments, ./
BUILD/DISCO_F469NI/GCC_ARM-RELEASE_O3/kws_realtime_test.elf does not
c:/users/admin/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-e
abi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific
data of file ./STM32_Audio/Addons/PDM/Lib\libPDMFilter_CM4_GCC.a(pdm_filter.o)
collect2.exe: error: ld returned 1 exit status
Let me know, if any one got solution for this ?

houxd, 2018-12-06:
ARM's official toolchain can solve this problem. I just encountered the same problem. I used STM32CubeMX to generate Makefile compilation on ubuntu18.04. The toolchain comes from gcc-arm-none-eabi installed by ubuntu apt. The solution is first Uninstall the original toolchain:
Sudo apt remote gcc-arm-none-eabi binutils-arm-none-eabi
Then install the arm official toolchain:
Sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
Sudo apt-get update
Sudo apt-get install gcc-arm-embedded
Please refer to:
Https://developer.arm.com/open-source/gnu-toolchain/gnu-rm
Although windows I have not tested, but I think the same can be solved

dientc, 2019-01-30:
> ARM's official toolchain can solve this problem. I just encountered the same problem. I used STM32CubeMX to generate Makefile compilation on ubuntu18.04. The toolchain comes from gcc-arm-none-eabi installed by ubuntu apt. The solution is first Uninstall the original toolchain:
> Sudo apt remote gcc-arm-none-eabi binutils-arm-none-eabi
> Then install the arm official toolchain:
> Sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
> Sudo apt-get update
> Sudo apt-get install gcc-arm-embedded
> Please refer to:
> [Https://developer.arm.com/open-source/gnu-toolchain/gnu-rm](Https://developer.arm.com/open-source/gnu-toolchain/gnu-rm)
> Although windows I have not tested, but I think the same can be solved
This link may help:
https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa

akohlsmith, 2019-08-14:
This is still an issue in August 2019 with Ubuntu 19.04; is there a reason that the official embedded arm packages don't correctly work with hardfp?

Gohith, 2020-09-13:
> When I compile default example by "make BOARD=stmf32f4discovery"
>
> ```
> usr/local/bin/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: [...]

Simply clean project and build it again ; Give it a try