RIOT-OS/RIOT #2664: internal error: _native_handle_tap_input

cgundogan, 2015-03-20:
I am facing this error when using the rpl_udp example in native.
I guess it might be due to my system upgrade.. I upgraded linux (3.18.6-1 -> 3.19.2-1).
When I start a node it crashes (somewhat randomly).
Is anyone else seeing this?

gebart, 2015-03-20:
Did you try running it in gdb?

cgundogan, 2015-03-20:
Not that much, I can say that `real_read` is returning 0 bytes, which in case results in the mentioned error message.
https://github.com/RIOT-OS/RIOT/blob/master/cpu/native/net/tap.c#L81

LudwigOrtmann, 2015-03-21:
My up-to-date arch does not have problems.

gebart, 2015-03-21:
I'm using a 3.19.0 kernel on Gentoo and I have not seen this problem before.

LudwigOrtmann, 2015-03-21:
@cgundogan Maybe you have some network manager or conflicting network tool (e.g. vpn tunnel) or permission management tool installed?

cgundogan, 2015-03-22:
@LudwigOrtmann I am not that sure about this as `real_read()` only returns sporadically `0`, so the read operation does work actually.
When I remove the `0` case, native works for me again. See #2686

cgundogan, 2015-04-07:
fixed by #2686 , so I am closing this issue.

closed 2015-04-07 by cgundogan