RIOT-OS/RIOT #2691: thread

gedw99, 2015-03-23:
this is google new IOT protocol over bluetooth. its what the Nest device runs.

http://threadgroup.org/Home.aspx

i was using zigbee. Oh what a mess...

i am very happy about riot OS.

i am wondering if this is a good protocol to work into the stack ?

what do people think ?

OlegHahm, 2015-03-23:
Hi Ged and welcome to RIOT! Usually this type of discussion is better done on one of the mailing lists (devel or users, see https://lists.riot-os.org/). Actually, a discussion about consortia like this (thread is luckily not a Google-only thing) has recently be started on the devel mailing list: https://lists.riot-os.org/pipermail/devel/2015-March/002282.html
Feel free to subscribe and participate there!

closed 2015-03-23 by OlegHahm

gedw99, 2015-03-23:
Hey okeg,

Yes I understand. Will converse on ideas on the lists.

Really happy I found this project.